PX4/PX4-Autopilot #14214: Preflight check fail Magnetometer compass reading inconsistent mRO Pixracer

sritank, 2020-02-23:
Hi,
I am using the mRO Pixracer board for my drone autopilot and the Taoglas sensor as my onboard GPS sensor. I calibrated my compasses based on the PX4 guide [here](https://docs.px4.io/v1.9.0/en/config/compass.html) . 

The three onboard Magnetometers have parameter names: 
MAG0 - Taoglas compass 1 = External Compass Orientation
MAG1 - Taoglas compass 2 = External Compass Orientation 1
MAG2 - Pixracer's compass = Autopilot Compass orientation

Currently the "heading" arrows on both the boards (Pixracer and Taoglas GPS) point forward, i.e. direction of flight. Hence the sensor orientation for the autopilot was set to "ROTATION_NONE". Affer successfully calibrating, the orientation for the other two compasses were also set to "ROTATION_NONE". However when I try to arm them, QGroundControl (QGC) gives an error that the compasses are inconsistent. So based on this [post](https://discuss.px4.io/t/gps-orientation-pixracer-mag-sensors-inconsistent/6058) , I found that the Taoglas compasses are oriented 90 deg away from the heading arrow. Even after accounting for this and setting the calibration sensor orientation to YAW_90 and YAW_180, I still get the same error.

I disabled MAG0 compass and was using only MAG1(Taoglas) and MAG2(Pixracer). Upon doing this I get a different preflight failure,i.e. "Yaw estimate error". 

Ultimately I disabled MAG0 and MAG2 sensor and am able to arm the drone using only the MAG1 sensor enabled.

Question: Must I account for something else while calibrating the sensors? And, how do I find out which sensor is the root cause of the above inconsistency?

sritank, 2020-02-28:
Any help with the above would be appreciated

julianoes, 2020-03-02:
@dagar you mentioned something about this, right?

julianoes, 2020-06-02:
@dagar sounds familiar?

dagar, 2020-06-02:
> @dagar sounds familiar?

Unfortunately. @sritank can you post a photo (or link) of the exact GPS/compass unit you're using?

A flight log from boot (SDLOG_MODE 1) or the output of `listener sensor_mag` would also help.

sritank, 2020-06-16:
I thought of replying after getting the flight log that you had requested. However, I don't know when I'll be allowed to go back and get access to the hardware. The compass I'm using is below:

https://store.mrobotics.io/ProductDetails.asp?ProductCode=mro-gps003-mr