PX4/PX4-Autopilot #14223: EKF does not switch to secondary GPS if main fails

CarlOlsson, 2020-02-24:
**Describe the bug**
If a vehicle has two GPS modules connected and the main one fails while it has a higher fix type than the secondary GPS the EKF will not switch to only use the secondary GPS but stop fusing GPS completely.

**To Reproduce**
Collect a replay logfile with a vehicle using two GPS modules. Stop publishing of the first GPS module when it has a higher fix than the secondary GPS

[image]

**Expected behavior**
The EKF should switch to only use the secondary GPS

**Additional context**
https://github.com/PX4/Firmware/blob/f3fefd7d9372ec98d404e2e99da2e251114a2d7e/src/modules/ekf2/ekf2_main.cpp#L953-L988

The problem is that `_gps_state[0].fix_type` is never updated after the GPS failure so `_gps_select_index` will never be set to one

CarlOlsson, 2020-02-24:
It also happens the other way around, and even if the GPS modules have the same fix type

julianoes, 2020-03-02:
FYI @DanielePettenuzzo

CarlOlsson, 2020-03-10:
solved by https://github.com/PX4/Firmware/pull/14278

closed 2020-03-10 by CarlOlsson