PX4/PX4-Autopilot #14232: EKF2: GPS blending offset not decreased to zero if one modle lose fix

CarlOlsson, 2020-02-25:
**Describe the bug**
When using GPS blending we calculate a position offset to apply to each measurement in order to smooth out the data we push into the EKF in case of fast varying GPS accuracy metrics. However, if one GPS module lose fix we don't decrease this value but keep applying the last calculated offset indefinetly.

**To Reproduce**
Collect a replay logfile with two GPS modules connected. During replay, force one GPS module to lose fix.

**Expected behavior**
The GPS data we push into the EKF should converge to the GPS module still working if the other lose fix.

**Log Files and Screenshots**
A horizontal position offset is added to one GPS module to more easy illustrate the issue. At t=100s the main GPS lose fix.
[image]

julianoes, 2020-03-02:
@DanielePettenuzzo is this the same that you found?

DanielePettenuzzo, 2020-03-02:
I'm not sure. They could be related

CarlOlsson, 2020-03-10:
solved by https://github.com/PX4/Firmware/pull/14278

closed 2020-03-10 by CarlOlsson