PX4/PX4-Autopilot #14251: EKF2: GPS blending seems to give wrong output for fast changing position meas.

CarlOlsson, 2020-02-27:
**Describe the bug**
If the position measurement of one GPS module changes fast the output of the blending algorithm gives a faulty output

**To Reproduce**
Here is a replay logfile with two GPS modules connected. At t=80s a position offset is added to gps0 (blue). The red line is the internal GPS blended state and the green line is the output of the blending algorithm that we push into the ECL
[image]

**Expected behavior**
The offset between the output signal and the internal blended gps state should be monotonically decreasing.

**Additional context**
The issue seems to originate from the fact that we account for the blending weights twice, once in the calculation of the offset between the _gps_output[0-1] and the _gps_blended_state, and once in the offset between the _gps_output[2] and the _gps_blended_state

bresch, 2020-02-27:
FYI @DanielePettenuzzo

CarlOlsson, 2020-03-10:
solved by https://github.com/PX4/Firmware/pull/14278

closed 2020-03-10 by CarlOlsson