PX4/PX4-Autopilot #14260: EKF2 Replay drops "multi-rotor drag fusion"

thomaswatters, 2020-02-28:
Seen on version: v1.9.2

When attempting to replay the attached log in replay_mode=ekf2 the drag_innovations are dropped (DNE), and the EKF2_AID_MASK appears to be changed to 0b1.

Exporting the parameters in the attached log show EF2_AID_MASK to be 0b10001 but exporting the parameters from the produced log (made by replay) shows EF2_AID_MASK to be 0b1.

Steps to reproduce the behavior:
1. Export parameters from attached file
2. Setup environment for ekf2 replay 
3. run px4_sitl none (log file is relatively long ... sorry)
4. examine generated log file 
5. Expect drag innovations to exist. Actual drag innovations are zeroed
6. Export parameters from generated log file
7. Expect EKF2_AID_MASK to be 33, Actual EKF2_AID_MASK 1

[log_27_2020-2-27-14-45-24.ulg.zip](https://github.com/PX4/Firmware/files/4267716/log_27_2020-2-27-14-45-24.ulg.zip)

vanurag, 2020-03-24:
I noticed similar issue with several parameters not matching between the original log file and the log file generated after replay. I fixed this issue by adding 'replay tryapplyparams' in rc.replay before 'ekf2 start'.

bkueng, 2020-03-25:
Thanks for looking into it.
Can you send a pull request?
The problem is that ekf does not take these params after it's started.

closed 2020-06-25 by bresch