PX4/PX4-Autopilot #14262: Runway takeoff not tracking

ryanjAA, 2020-02-29:
Runway takeoff (on fixed wing with the nose gear on its own channel) is not tracking. It immediately veers off to the right substantially taking it off the runway. 

3 logs of this with one log having large gains and unlimited travel for good measure on the wheel controller.

To further test this we took another aircraft in a different location and recreated. That log is not attached but it obeyed in the identical way.

There is more than enough wheel authority, no errors, mag issues, etc so understanding the lack of tracking and any mission start yields to veering off to the right would be important to understand.

https://review.px4.io/plot_app?log=d7858c28-bb1f-49df-9df5-c8fd04755a80
(vibration only because it was taken at high speed off the runway)

https://review.px4.io/plot_app?log=0d6e9b1f-1d37-43fb-a673-c19de0298fac
(Immediate right upon initiating the mission - only look at the mission portion of the log)

https://review.px4.io/plot_app?log=65c49e9f-61a4-4a75-8813-4ce66160a4c4
(Immediate right upon initiating the mission - only look at the mission portion of the log but with larger wheel rate gains for good measure.)

It would seem like something is wrong for it to always want to go right first. Is this a bug or something that simply now needs to be re-tuned so there is a much more subdued effect on the WR controller's abilities? I just find it strange since this did occur occasionally over the years but not every time which is happening now.

jinchengde, 2020-02-29:
have you notice that throttle ramp up is fast and reverse twist from propeller is too large?
https://github.com/PX4/Firmware/issues/13146

ryanjAA, 2020-02-29:
No, since the nose servo (output 5 in the log) immediately has a large increase and sometimes maxes out when throttle ramp up increases. Plus I’d of expected to see this a few years ago so it looks like it’s purposely steering it off to the right. 

No mag issues, mag ratio is 0. No checkbits either.

ryanjAA, 2020-03-01:
Collecting more logs of this issue on different systems.

jinchengde, 2020-03-02:
Is that still exist in master ?

ryanjAA, 2020-03-03:
Yes. This is still in master. I’m actually wondering if this is something on the math side calculating heading since if we start looking at batches of recent logs on different aircraft, there is almost always an immediate rise in the pwm value of the nose wheel. I opened up some logs of another aircraft type that had exactly the same issue.  Oscillation is present until speed is achieved but all go to the right off course. In the first log above I even made the waypoint cheated to the left so the plane would need to turn left slightly and it still went hard right.

jinchengde, 2020-03-07:
maybe you could check your airspeed sensor first.
look at your flight log for airspeed
[image]
I suppose the vehicle can't reach the airspeed setpoint, so stay at the stage of CLAMPED_TO_RUNWAY but can't reach TAKEOFF, but in that time, vehicle had liftoff, so the gear can not control the yaw

ryanjAA, 2020-03-07:
Ground based speed when done in manual or stabilized can go quite a bit faster without issue. Takeoff speed is set to just above this. It’s changed out of mission because it’s too close to leaving the side of the runway. The issue is actually right when mission is initiated and you can see the wheel output go rapid. These planes have all flown many many times without issue. I’m starting to think this how the wheel rate controller is setup although why a sudden change is output upon mission initiation also doesn't make sense. We’re going to retune them with very dampened settings which will likely work as a work around but not be ideal. It’s definitely not hardware though and never used to happen before and it’s not just on these specific aircraft, some others we tested and took logs from as well, even other types of aircraft completely all see the same thing.