PX4/PX4-Autopilot #14273: ESC calibration broken after 1.10?

julianoes, 2020-03-02:
Is ESC calibration still working with 1.10 and later?

This post suggests otherwise:
https://discuss.px4.io/t/unable-to-start-esc-calibration/15033

tacdin, 2020-04-10:
Do you have a solution for this topic ? I have the same problem.

tiralonghipol, 2020-04-13:
hi i had the same issue
rolling back to 1.9.x solved the problem @tacdin

fabrizioschiano, 2020-09-24:
Did someone have any luck reliably calibrating ESCs with firmware 1.10 and later?

dagar, 2020-09-24:
> Did someone have any luck reliably calibrating ESCs with firmware 1.10 and later?

Are you seeing any particular error?

tiralonghipol, 2020-09-24:
@fabrizioschiano  I flashed v1.11 on pixracer (px4_fmu_v4) last night and works perfectly
I was able to kind of reproduce the behavior when downloading the firmware from QGC vs. flash from source/terminal

In the second case I did not have that problem

dagar, 2020-09-24:
> I was able to kind of reproduce the behavior when downloading the firmware from QGC vs. flash from source/terminal

Can you elaborate? Are you saying if you flash from QGC then ESC calibration fails?

coderkalyan, 2021-01-08:
I've been able to calibrate ESCs (at least, I'm not getting a "Transition denied: SHUTDOWN to INIT" error) on 1.11 so this issue is probably not universal.

julianoes, 2021-06-28:
I'm going to close this because this looks stale and neither do I see any actionable notes.

closed 2021-06-28 by julianoes