PX4/PX4-Autopilot #14274: Stm32H7 hard fault  when SPI2 and I2C4 work together

zhaoxiaowei1013, 2020-03-02:
**Describe the bug**
I have a stm32H7 hardware ，drive barometer using SPI2 interface,and driver magnetometer（ ist8308 ）using I2C4 and I2C3  interface，when I run both drivers in the work queue，the system will generate hard fault.
I am according to the documentation to locate where the hard falut happened，the hard fault happened in  spi read when it be called in work quene transfer.

**To Reproduce**

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
[fault log ](https://drive.google.com/open?id=1x8EJ7IroYgYd6fPogqPMdqiSsHCVQbqE)
[elf file] (https://drive.google.com/open?id=1keEE1-alq0i1zDbRLMBzWQZS6ixmTBaW)
[debug uart console] (https://drive.google.com/open?id=14WDtZTGo_W4fwkI7egY3H61Nn85a-xnu)
```
(gdb) info line *0x080d84b9
Line 134 of "../../src/lib/drivers/device/nuttx/SPI.cpp"
   starts at address 0x80d84b4 <device::SPI::transfer(unsigned char*, unsigned char*, unsigned int)+18>
   and ends at 0x80d84bc <device::SPI::transfer(unsigned char*, unsigned char*, unsigned int)+26>.
(gdb) info line *0x080d84fe
Line 147 of "../../src/lib/drivers/device/nuttx/SPI.cpp"
   starts at address 0x80d84fa <device::SPI::transfer(unsigned char*, unsigned char*, unsigned int)+88>
   and ends at 0x80d8502 <device::SPI::transfer(unsigned char*, unsigned char*, unsigned int)+96>.
```

and I can push my code if necessary。

julianoes, 2020-03-02:
How can anyone reproduce this to fix it?

zhaoxiaowei1013, 2020-03-02:
my imu bmi088 in SPI4 and baro（tcbp001ta）in SPI2 
[image]
and I2C3 is external magnetometer（ist8308）  I2C4 is internal magnetometer （ist8308）
[image]
Just can't work when SPI2 and I2C3 or I2C4 together。

dagar, 2020-03-02:
If you open pull requests for the IST8303 magnetometer or tcbp001ta (model?) barometer we could take a look and maintain them upstream, otherwise there's not much we can do here.

There shouldn't be any overlap between these things unless there's an error. Each workqueue is a separate thread, so all the activity, locking, etc should be independent.

zhaoxiaowei1013, 2020-03-03:
please view the pr [#14286](https://github.com/PX4/Firmware/pull/14286)。
thank you very much！

zhaoxiaowei1013, 2020-04-24:
in the master branch，this problem is fixed

closed 2020-04-24 by zhaoxiaowei1013