PX4/PX4-Autopilot #14277: Packet Loss Calculation

0xF0F0F0F0, 2020-03-02:
Hi,

I wanted to know if there is an accurate way to calculate packet loss on a mavlink channel. I have come across calculation using message sequence number but i have a few questions:

1. What is the difference between seq number in mavlink packet vs. current_rx_seq in mavlink_status?
2.  If i lose more than 255 sequence numbers, will it affect my calculation? (since seq num is 0-255)

hamishwillee, 2020-03-04:
FYI Duplicate in mavlink/mavlink: https://github.com/mavlink/mavlink/issues/1315

hamishwillee, 2020-03-30:
@julianoes Who is the right person to help progress this one?

julianoes, 2020-03-30:
You? Or the question asker? Not me probably.

hamishwillee, 2020-03-30:
@0xF0F0F0F0 Are you still interested in this? If so I can apply my brain to it shortly.

hamishwillee, 2020-04-07:
Closing. Assume poster has lost interest. Also it is a duplicate. Reopen one of them if you need to do more with this.

closed 2020-04-07 by hamishwillee