PX4/PX4-Autopilot #14281: HITL Missing battery parameters

depauwjimmy, 2020-03-02:
I have been struggling for the past week to use a simulator but without any results so far.
What i wish to achieve is :
- HITL on a Cube with 1.10.1
- AirSim
- Xbox controller for manual control

Since SITL does not work at all with AirSim i wanted to try HITL but i have a weird problem.
Fresh flashed system show missing parameters error.

BAT_V_CHARGED, BAT_N_CELLS and BAT_V_EMPTY

Since this is as a preflight requirements i just can't do anything.
This is exactly the same issue as https://discuss.px4.io/t/missing-parameters-on-switching-to-hitl-with-pixhawk4-on-px4-v1-10-0/14423 but this has no answers.

To reproduce :
- Fresh flash a Pixhawk2.1 Cube with 1.10.1
- Select HIL Quadcopter X
- Reboot

julianoes, 2020-03-03:
You should be able to work around the preflight errors using:
https://dev.px4.io/master/en/advanced/parameter_reference.html#CBRK_SUPPLY_CHK

depauwjimmy, 2020-03-04:
Thanks i have managed to start both AirSim and QCG using 2 usb connection which allows me to use a Pad for manual controls.
Now i have other problems :/

My biggest issue right now is that it always wants to land and never hover when i release the controls.
Is COM_OBL_ACT not supposed to prevent it?
I also get a random 

    EMERGENCY: Critical navigation failure! Check sensor calibra

This happen quickly after takeoff and with the forced landing always there it is simply impossible to control.
Tried with both 1.9.2 and 1.10.1 and behavior is the same.

rickyjames35, 2020-05-17:
[image]
I can reproduce this exact bug when using the HITL. I also get this notification when going to "Flight Modes"

julianoes, 2020-05-18:
> My biggest issue right now is that it always wants to land and never hover when i release the controls.

Do you have a log that shows that?

jasonotu, 2020-06-01:
I'm having the same issue.

dagar, 2020-08-31:
This has been fixed.

closed 2020-08-31 by dagar

McKayCyborg, 2020-10-12:
It appears that this issue is still present in the Current PX4 firmware build. I am trying to run HITL using AirSim on a Pixhawk 4 board and I am unable to run a mission due to the following errors:

When I boot up, it gets this issue with battery parameters:
[image]

Then, when trying to upload a mission, it gets these issues:
[image]

[image]

Also, I get a message saying that the vehicle requires setup prior to flight.

[image]

Current setup:
QGC Version - v4.0.10
PX4 Firmware version - v1.11.1

What I have tried:
1) Re-flashing the board
2) Re-installing QGC
3) Resetting all parameters on the board
4) Setting the CBRK_SUPPLY_CHK to 894281

Nothing appears to work and I am unable to fly a mission. I am able to arm and fly it manually using an RC Radio controller, however.

Has this issue actually been fixed?