PX4/PX4-Autopilot #14287: Packet drop while using mavlink forwarding in px4 mavlink module

myeabhishek, 2020-03-03:
Hello Team,

I am using Mavlink forwarding to route mavlink messages between Telem1 and Telem 2 ports. I have observed on sending high-frequency data on telem1 i.e. 400 bytes/second, I am getting about 10% packet drop on telem2. This increases linearly with increase in frequency of data. There is no other interference in the system. I am able to receive 100 percent of the data on other hardwares. This is not a communication/RF link issue. 

I can see that there are alot of parameters in PX4 controlling the message buffer size and data rate. How can I tune these parameters to get best results for my application?

Following are my settings in RCS script:
Telem 1: MAVLINK_F "-r 1200 -f -b 57600
Telem 2: 
 mavlink start -d ${MAVLINK_COMPANION_DEVICE} -b 57600 -m onboard -r 5000 -x -f

julianoes, 2020-03-03:
Can you provide the output of `mavlink status` for the ok and no-ok case?

myeabhishek, 2020-03-03:
Hi, 

I have attached the output of mavlink status command for both cases. Thank you for your assistance.

[image]

[image]

julianoes, 2020-03-03:
I was expecting to see a `rate mult` of 1 for the working case and a number smaller than 1 for the not ok case. Instead the `rate mult` seems the same for both cases. However, the ping statistics seem bogus but that might not matter.

Have you tried choosing a higher `-r ` rate for the companion side? Basically, the `rate mult` is < 1 if the asked rates of all the messages is too much compared to the available rate (`-r`).

myeabhishek, 2020-03-03:
Yes, I increased the -r for companion link to 10000. The packet loss increased from 10 % to 45 %. 

Is serial baud rate related to these configurations? I have set 57600 for both ports.

myeabhishek, 2020-03-03:
Following is the screenshot for mavlink status. I increased the baud rate for telem2 to 921600 with 10000 as my -r value. 

My packet loss has decreased from 45% in last attempt to 15% and rate multiplier is also set to 1 now. 

[image]

Jackiemesser, 2021-01-03:
Hey, did you ever find a solution to your problem? I'm having a similar issue

USS-ICI, 2023-10-05:
So my code works for Ardupilot, but drops packets continually with PX4 Firmware, did ya'll find a fix?

julianoes, 2023-10-05:
Fix coming #21074