PX4/PX4-Autopilot #14291: Detailed logs in gazebo+jmavsim for simulated flight

Sneha-hub115, 2020-03-04:
Hi,
I'm using ubuntu18.04 to build the px4 code for gazebo and jmavsim. I'm stuck with the following issues:
1. How to get detailed logs in gazebo+jmavsim for simulated flights.
2. How to do diagnostic logging.
2. How to get detailed logs in QGC+real drone flight.

Thank you.

julianoes, 2020-03-04:
Hi, you need to be bit more specific. What do you mean with detailed logs?

In general a question like this is better to be posted in discuss.px4.io since it's not a specific issue with the PX4 firmware.

Sneha-hub115, 2020-03-04:
Hi, I want to know where and how to get log files in gazebo and jmavsim using px4 code.

Thank you

julianoes, 2020-03-04:
The ulogs are stored in `Firmware/build/px4_sitl_default/tmp/rootfs/log/...`. 

For real flights the logs are stored on the MicroSD card and you can download them using QGroundControl.

Also, see https://dev.px4.io/master/en/log/flight_log_analysis.html.

closed 2020-03-04 by julianoes