PX4/PX4-Autopilot #14292: Add AGL altitude (distance sensor) for quadchute minimum alt

moreba1, 2020-03-04:
Hi

i think AGL alt is better for quadchute minimim alt than relative alt . if we choice AGL altitude for quadchute we can prevent collision with ground .

julianoes, 2020-03-04:
Please fill out the issue template and give enough context, code, logs etc.

moreba1, 2020-03-05:
VT_FW_MIN_ALT (FLOAT)
**QuadChute Altitude**
Comment: Minimum altitude for fixed wing flight, when in fixed wing the altitude drops below this altitude the vehicle will transition back to MC mode and enter failsafe RTL

this altitude is relative altitude but Relative altitude is not important, because It may not be flat ground and altitude above terrain is important. for measuring we can use distance sensor and 
https://docs.px4.io/v1.9.0/en/sensor/rangefinders.html
DISTANCE_SENSOR.current_distance

[image]

julianoes, 2020-03-05:
Ok, and what's your suggested change? I'm not entirely sure on that yet.

moreba1, 2020-03-05:
What is @sanderux  opinion? He developed the quadchute code .

sanderux, 2020-03-05:
QuadChute has 2 modes:
VT_FW_MIN_ALT defining a minimum altitude above home
VT_FW_ALT_ERR defining a maximum deviation from the altitude setpoint.

Using range finders for quadchute would be a bad idea because;
1. Range finders generally dont operate at the altitude where you would want quadchute to start. 
2. Quadchute engages when the vehicle is not flying properly, a rangefinder needs to be level to give a good estimation and that is generally not the case when a quadchute event is required.

I recommend using the VT_FW_ALT_ERR param to achieve terrain aware failsafe conditions.

moreba1, 2020-03-05:
> Using range finders for quadchute would be a bad idea because;

what is your opinion in adding range finder alt in addition to Rel alt not replacing with it ?

this would be useful when terrain alt has variance with reality or there is some building.

sanderux, 2020-03-21:
I dont see the benefit above using VT_FW_ALT_ERR. 
That would normally  follow terrain too

closed 2020-12-26 by moreba1