PX4/PX4-Autopilot #14293: lock CONS debug console  when disconnecting usb cable after usb connection

stmoon, 2020-03-04:
**Describe the bug**

When disconnecting USB cable after the connection, the CONS debug console is locked.
When I connect the USB cable again, the CONS debug console is released. 

In this case, some tasks are held until connecting USB cable again. 

I think it is related to mavlink.

julianoes, 2020-03-05:
What's the "CONS debug console"?

stmoon, 2020-03-06:
Hi, 

In the Pixhawk 2.1, there is CONS (console) near the SBUS.

[image]

julianoes, 2020-03-06:
Aha! @davids5 do you know?

davids5, 2020-03-06:
@stmoon What else do you have connected?

stmoon, 2020-03-06:
I just connected POWER, CONS, and usb (ttyACM) for test.

I use master version firmware and Pixhawk 2.1.

[image]