PX4/PX4-Autopilot #14300: SITL Gazebo with optical flow not working

julianoes, 2020-03-05:
**Describe the bug**

I can't switch into position control. Only altitude control works, even if in-air.

Sometimes the flow thing reports matches, but even then I can't switch into position control:
```
Number of good matches: 4, desired: 3
Number of good matches: 3, desired: 3
Number of good matches: 2, desired: 1
```

However, most of the time, this is not reporting anyway.

The LOCAL_POSITION_NED velocity estimates seem correct but position is just 0.

**To Reproduce**
Steps to reproduce the behavior:
```
make px4_sitl gazebo_iris_opt_flow
```
Then try to switch to position control, or switch to altitude control to takeoff and then try to switch to position control. Both don't work for me.

**Expected behavior**
I assume position control should work, especially when there is also GPS being sent which is right now the case.

Jaeyoung-Lim, 2020-05-22:
@julianoes Is this still an issue? I can fly with the iris_opt_flow model

julianoes, 2020-05-25:
@Jaeyoung-Lim I just tried again doing a takeoff in QGC with `iris_opt_flow` and it flipped. This is with Gazebo 11.

https://logs.px4.io/plot_app?log=aa17113c-b016-4118-a14c-b5c63bca6281

julianoes, 2020-05-26:
Turns out this works now, even if I disable GPS publication.

closed 2020-05-26 by julianoes