PX4/PX4-Autopilot #14303: SITL jmavsim crazy pitch rate oscillations

MaEtUgR, 2020-03-05:
**To Reproduce**
Steps to reproduce the behavior:
1. master on this commit https://github.com/PX4/Firmware/commit/2b087a92c277d542405c79fd98e99e5ccfb358ca
2. run `make px4_sitl jmavsim`
3. immediately switch to altitude mode and takeoff by pushing up with the stick
4. Don't move the sticks anymore and see the drone having fast and extreme pitch oscillation for multiple seconds

**Expected behavior**
There used to be stable flight from the beginning on.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=44c727a2-9b7c-43e7-b86d-f4d58e5e50d3
[image]

**Additional context**
- Suspect might be https://github.com/PX4/Firmware/pull/14228 (NOT YET VERIFIED)
I'll do some further testing to narrow it down.

FYI @TSC21 @Jaeyoung-Lim

MaEtUgR, 2020-03-05:
I bisected it to come from the vehicle_angular_velocity filtering:
https://github.com/PX4/Firmware/commit/24f0c2d72ae2837c5d41d6835f8d18ea1647742e#diff-e1d4d75170cc48374d5f144ebfca43d9
The oscillation only happens shortly after boot and it stops exactly when the message 
`INFO  [vehicle_angular_velocity] updating filter, sample rate: 1000.000 Hz -> 250.000 Hz`
is displayed coming from https://github.com/PX4/Firmware/commit/24f0c2d72ae2837c5d41d6835f8d18ea1647742e#diff-e1d4d75170cc48374d5f144ebfca43d9R112

MaEtUgR, 2020-03-05:
@dagar Not sure if this is a crucial real world issue since in real world you usually don't take off that fast but it's still undesired and we should try to get rid of the delay. How did it work before?

FYI @bkueng

dagar, 2020-03-05:
> @dagar Not sure if this is a crucial real world issue since in real world you usually don't take off that fast but it's still undesired and we should try to get rid of the delay. How did it work before?
> 
> FYI @bkueng

There have been a few iterations in the last year, but at one point the LPF was in the sim driver itself (sometimes with an incorrect sample rate), then it moved to PX4Accelerometer/PX4Gyroscope where the driver was setting the sample rate (again incorrectly in many cases), then it finally moved to sensors/vehicle_angular_velocity where we started measuring the sample rate ourselves.

We need to either set a more appropriate default value or update it sooner (or both). Having the simulator provide sensor data closer to a real system (1 kHz instead of 250 Hz) would also be nice, but I'm not sure if that's realistic and it would break the recent lockstep ekf2_timestamp fixes.

https://github.com/PX4/Firmware/blob/b9fab04adbfd92783206a08364c8fea077f1ed39/src/modules/sensors/vehicle_angular_velocity/VehicleAngularVelocity.hpp#L103

Note: mc_rate_control is also doing something similar to get the sample rate. This would go away with https://github.com/PX4/Firmware/pull/14048. 
https://github.com/PX4/Firmware/blob/b9fab04adbfd92783206a08364c8fea077f1ed39/src/modules/mc_rate_control/MulticopterRateControl.cpp#L243-L255

LucaMex, 2020-03-11:
I don't know if that might be useful, but I've noticed (running separately px4_sitl and jmavsim) that the error occurs only (or mainly?) when the lockstep isn't used

TSC21, 2020-03-12:
This is a real issue not only in jMAVSim but also in Gazebo. It can also be reproduced in the Solo for example.

MaEtUgR, 2020-03-12:
I was running lockstep with the normal linux setup.

PR #14622 (2020-04-11): sensors/vehicle_angular_velocity: check filter based on time threshold instead of samples
With the filter check based on a certain number of samples it takes 4 times longer (or more) in SITL (250 Hz IMU) vs a real vehicle (1 or 2 kHz IMU). 

 - fixes #14303
commit 7b4f6b69 (2020-04-11): sensors/vehicle_angular_velocity: check filter based on time threshold instead of samples

 - fixes #14303
closed 2020-04-11 by dagar