PX4/PX4-Autopilot #14309: SITL GAZEBO fails to run on

mhk4zemi, 2020-03-06:
**Describe the bug**
I am using windows subsystem for Linux, installed ubuntu 18 on the windows power shell and also I have installed gazebo and firmware without any error. 
Once I try to run the simulation ''make px4_sitl gazebo'' I will get the following error. 

```
make px4_sitl gazebo
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0")
-- Boost version: 1.65.1
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
[... 101 lines cut ...]
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
[Wrn] [GuiIface.cc:119] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-hossein'
[Wrn] [GuiIface.cc:119] failed to get the current screen resources
Error: no autostart file found (etc/init.d-posix/ERROR [px4] PX4 daemon not running yet_*)
ERROR [px4] Startup script returned with return value: 256
pxh> [Wrn] [GuiIface.cc:119] QXcbConnection: XCB error: 170 (Unknown), sequence: 170, resource id: 90, major code: 146 (Unknown), minor code: 20

```

hamishwillee, 2020-03-06:
Windows subsystem for Linux is not supported by Gazebo. This has never been known to work.

You may be able to get JMAVSim working. Generally I recommend the Cygwin solution for jMAVSim, and a VM with Ubuntu if you really must have Gazebo.

closed 2020-03-06 by hamishwillee

mhk4zemi, 2020-03-06:
I found the fix at this [commit](https://github.com/PX4/Firmware/commit/01818b505f898f33176bf90f9563e84942ea56cf)

hamishwillee, 2020-03-06:
So you got gazebo running under WSD? 
Well done.

mhk4zemi, 2020-03-06:
yes

hamishwillee, 2020-03-09:
Thanks. Actually we did get this working once (I had forgotten). However it was too slow to be useful except when run headless: https://dev.px4.io/master/en/setup/dev_env_windows_bash_on_win.html#simulation-jmavsim

mhk4zemi, 2020-03-09:
I manage to get the SITL-GAZEBO to work with a realtime factor of 1. for the GUI I used Xming.

LorenzMeier, 2020-03-09:
Could you send a pull request?

mhk4zemi, 2020-03-10:
Sure I will do it on the weekend

bamboosdu, 2020-09-29:
What is the wrong with "ERROR [px4] PX4 daemon not running yet"?
I met the same problem.