PX4/PX4-Autopilot #14313: [fixed wing]nose down after mission mode

jinchengde, 2020-03-06:
**Describe the bug**
When I switch the flight mode from stablize to mission, the plane have a strange node down and power max immidately, I haven't see that before, and the speed is enough to level flight, but I think maybe some problem about airspeed because I haven't airspeed sensor, the speed is relate on the groundspeed from GPS.
the flight log is below:
https://logs.px4.io/plot_app?log=1a3116dd-d687-470f-8cb3-31740f9e0ac6
speed:
[image]

jinchengde, 2020-03-08:
I changed ASPD_PRIMARY from '''disable''' to '''Groundspeed minus windspeed''' and solve this issue.

jinchengde, 2020-03-08:
I have a little confused about airspeed setting, current I don't use the airspeed sensor and the airspeed is depend on the groundspeed from GPS(the engine is pull up so difficult to install pitot tube)
so how about the parameter:
FW_ASPD_MODE (NORMAL(use airspeed if avaliable))
ASPD_PRIMARY(Groundspeed minus windspeed)
ASPD_DO_CHECK(0)
CBRK_AIRSPD_CHK(162128)

LorenzMeier, 2020-03-08:
Could you send a PR for your fix?

jinchengde, 2020-03-09:
@LorenzMeier 
I just modify the parameter and solve the problem, but actually I don't know the root cuase for this issue, so that's why I wish to simulator the fixed wing without airspeed in gazebo.
If the plane haven't install airspeed sensor, no thread will public the topic of ```airspeed``` in uORB, but only ```airspeed_validated``` will be public to uORB by airspeed selector. I suppose that maybe some confuse byt these two topics.
but in gazebo, we can't simulator this scenario, does anyone could figure out how to modify the gazebo plugin to disable the airspeed sensor?