PX4/PX4-Autopilot #14321: Ekf2: mag bias learning. Bug or feature?

Slimercorp, 2020-03-07:
Please, to take attention on this if struct:
https://github.com/PX4/Firmware/blob/d7a9b123e6b84df4a7c01daaac64706c39abd3d7/src/modules/ekf2/ekf2_main.cpp#L1533

If conditions was met sometime, **_total_cal_time_us** increasing, okay, but when these conditions aren't met, for example, for 60 second due **control_status.flags.mag_3D == false**, and again control_status.flags.mag_3D == true, in this case **_total_cal_time_us** will get 60 seconds by **(now - _last_magcal_us)**. I think it's bug, because during 60 second mag bias estimation wasn't peformed and should not get in _total_cal_time_us. To fix bug I suggest this if struct:

```
// Check if conditions are OK for learning of magnetometer bias values
if (!_vehicle_land_detected.landed && // not on ground
 (_vehicle_status.arming_state == vehicle_status_s::ARMING_STATE_ARMED) && // vehicle is armed
!status.filter_fault_flags && // there are no filter faults
control_status.flags.mag_3D) { // the EKF is operating in the correct mode

	if (_last_magcal_us == 0) {
		_last_magcal_us = now;
	} else {
		_total_cal_time_us += now - _last_magcal_us;
		_last_magcal_us = now;
	}

} else if (status.filter_fault_flags != 0) {
// if a filter fault has occurred, assume previous learning was invalid and do not
// count it towards total learning time.
	_total_cal_time_us = 0;

	for (bool &cal_available : _valid_cal_available) {
		cal_available = false;
	}
} else {
	_last_magcal_us = now;
}
```
What do you think, @RomanBapst?

RomanBapst, 2020-03-09:
@Slimercorp Yes, I think your suggestion makes sense. Do you want to create a pull request and fix it? Otherwise I will do it.

Slimercorp, 2020-03-09:
@RomanBapst Please do it you.

closed 2020-03-10 by bresch