PX4/PX4-Autopilot #14323: Can't clone the Firmware with git clone https://github.com/PX4/Firmware.git --recursive

gitzhangcx, 2020-03-08:
when i run `git clone https://github.com/PX4/Firmware.git --recursive` to clone this project, i can't clone the folder "Firmware/Tools/sitl_gazebo",the error is followed:
 ```
Submodule path 'src/modules/micrortps_bridge/micro-CDR': checked out 'a4927629115de47b5545632b91ad17df55f2d20d'
Unable to checkout '8569aec5bb709acb2fec7ccb5a5a4405f0cdec2a' in submodule path 'Tools/sitl_gazebo'
```
How can i fix it ? Thanks a lot !

julianoes, 2020-03-26:
@Norton-z this is probably fixed by now. Can you check?

closed 2020-05-25 by julianoes

chasadch, 2024-11-24:
root@AsadArshad:~/PX4-Autopilot# git clone https://github.com/PX4/Firmware.git --recursive
Cloning into 'Firmware'...
remote: Enumerating objects: 488387, done.
remote: Counting objects: 100% (81/81), done.
remote: Compressing objects: 100% (67/67), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 1414 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

julianoes, 2024-11-25:
@chasadch your issue is different, likely a problem with your internet. And please don't necrobump old threads.

chasadch, 2024-11-30:
@julianoes sir please help me to solve this i have tied now ...   root@AsadArshad:~/PX4-Autopilot# make px4_fmu-v5_default
-- PX4_GIT_TAG: v1.16.0-alpha2-51-gb06ff99a3e
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3")
-- PX4 config file: /root/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- ROOT_PATH /fs/microsd
-- PARAM_FILE /fs/mtd_params
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/NuttX.git' into submodule path '/root/PX4-Autopilot/platforms/nuttx/NuttX/nuttx' failed
Failed to clone 'platforms/nuttx/NuttX/nuttx'. Retry scheduled