PX4/PX4-Autopilot #14325: NXP K66: System console drops parameters

LorenzMeier, 2020-03-08:
@igalloway @davidaroyer The K66 console drops characters, as demonstrated by this failed test:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/pr-mavlink-ftp-filenotfound/17/pipeline

[image]

The code calls "tests parameters" but what arrives at the system is "test parameters" (which is an invalid command, the "s" got dropped). @dagar We should consider to disable the K66 until this is resolved, because it will lead to a flaky QA experience.

We did need to enable RX DMA in the past to ensure that no characters from a serial buffer could be dropped.

dagar, 2020-03-09:
I need to piece together where we could be dropping things with the current architecture.

From `perf latency` (same nxp k66 on test rack) we can get a reasonable idea of the worst case interrupt latency.

[image]

jarivanewijk, 2020-03-11:
Thank you for reporting and thanks @dagar for the fix.

Our team in Eindhoven has to work at home because of the corona virus. I currently don't have an FMUK66 at home, but I can go get one later today. Let us know if there is anything we might be able to help with.

jarivanewijk, 2020-03-31:
@dagar @LorenzMeier Did I misunderstand that there was already a fix for this issue? #14351 was closed without merging.

davids5, 2020-03-31:
@jarivanewijk - i believe this was a misunderstanding that the Nuttx driver for the K66 supported DMA on the lpuart. It does not.  I will need to look at what is involved in adding it.

davids5, 2021-07-20:
LPUART Serial DMA added. Console https://github.com/PX4/PX4-Autopilot/pull/17956

closed 2021-07-20 by davids5