PX4/PX4-Autopilot #14327: Mission execution requires re-upload of mission between restarts

LorenzMeier, 2020-03-08:
There seems to be a condition somewhere that leads to successful arming, but not mission execution if the mission is not uploaded prior to execution, but is still on the vehicle from a previous boot. I suspect the culprit to be the initialization state of something in the navigator.

Antiheavy, 2020-03-08:
Are you seeing that in SITL or on a real vehicle?  I’ve noticed SITL sometimes has weird reset issues where the onboard mission doesn’t seem to initialize right upon restart.  Something about in a real vehicle the mission resets from the start but in SITL it gets stuck on whatever the waypoint was Last tracking from the previous SITL session. Not sure if that is related....

julianoes, 2020-03-09:
I have seen these effects where the vehicle (I think mostly VTOL) would not take off unless I re-uploaded the mission.