PX4/PX4-Autopilot #14336: UAVCAN DSDL submodule issue with new master branch vs legacy-v0

ARob109, 2020-03-10:
The UAVCAN DSDL repository https://github.com/UAVCAN/public_regulated_data_types/tree/master deprecated the DSDL submodule version linked to in the PX4/libuavcan "px4" branch to DSDL  branch "legacy-v0". This breaks syncing submodules when trying to build the PX4 firmware. 

The PX4/libuavcan "px4" branch is tracked to DSDL commit 192295c. This commit is not present in the new "master" branch for DSDL.

Error is:
Fetched in submodule path src/drivers/uavcan/libuavcan/dsdl but it did not contain 192295c... direct fetching of that commit failed
Failed to recurse into submodule path 'src/drivers/uavcan/libuavcan/dsdl'

Workaround is to name modify gitmodules within PX4/libuavcan to specify branch of dsdl submodule as "legacy-v0".

julianoes, 2020-03-26:
Is this still an issue?

TSC21, 2020-03-26:
@julianoes it probably still is. @ARob109 do you mind issuing a PR to PX4/libuavcan fixing that? Thanks!

ARob109, 2020-03-26:
@TSC21, PR issued to PX4/libuavcan. New to contributing, this was my first PR, so hopefully I did it correctly. Still also very new to submodules, but the PX4/Firmware will also need to be updated to point to the new PX4/libuavcan commit, correct?

TSC21, 2020-03-26:
> @TSC21, PR issued to PX4/libuavcan. New to contributing, this was my first PR, so hopefully I did it correctly. Still also very new to submodules, but the PX4/Firmware will also need to be updated to point to the new PX4/libuavcan commit, correct?

I left a comment. You need to rebase in master.

TSC21, 2020-03-28:
Solved in #14510.