PX4/PX4-Autopilot #14354: Fixing up support for mRo Control Zero F7

pkocmoud, 2020-03-10:
Hi Dan and David.

Setup:
Control Zero with USB cable. 
SDCard inserted

Steps to reproduce the issue:
Loaded http://px4-travis.s3.amazonaws.com/Firmware/stable/mro_ctrl-zero-f7_default.px4 and board boots up fine.
Connect via QGC
Set frame type to generic quad and reboot - board locks up in the middle of playing the start up tone.

Output from console:
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: MRO_CTRL_ZERO_F7
FW git-hash: e0f016c2b3db160284bbaaadaf70bef84592d81e
FW version: Release 1.10.1 (17433087)
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: 427238133be2b0ecd068a11e886ee8fdbc31f6dc
Build datetime: Jan 21 2020 19:04:10
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 00020000000032363435323751110037001a
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
WARN  [rgbled] no RGB led on bus #1
nsh: rgbled_ncp5623c: command not found
nsh: rgbled_pwm: command not found
ERROR [param] Parameter LIGHT_EN_BLINKM not found
Board sensors: /etc/init.d/rc.board_sensors
MPU6000 on SPI bus 1 at 0 (1000 KHz)
nsh: bmi088: command not found
nsh: dps310: command not found
ERROR [battery_status] no ADC found: /dev/adc0 (2)
ERROR [fmu] setting I2C clock failed
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS2
Starting MAVLink on /dev/ttyS0
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS0 @ 57600B
ERROR [mavlink] offboard mission init failed (-1)
ERROR [param] Parameter DSHOT_CONFIG not found
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter SENS_EN_PX4FLOW not found
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1

Notes:
The MCU on this board is actually the STM32F777IIK6

Suggestions?

dagar, 2020-03-10:
Have you tried the current state of master? 

http://px4-travis.s3.amazonaws.com/Firmware/master/mro_ctrl-zero-f7_default.px4

pkocmoud, 2020-03-10:
No but I will tonight.

Phil

http://px4-travis.s3.amazonaws.com/Firmware/master/mro_ctrl-zero-f7_default.px4

pkocmoud, 2020-03-11:
Master is better, no more lock up.

Console output:
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: MRO_CTRL_ZERO_F7
FW git-hash: ab3670e02b195fbe477e6737138419ac51adfbab
FW version: 1.11.0 80 (17498240)
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: bf646dcf46ac4a18c29b1cab4aa5e7a54280782e
Build datetime: Mar 10 2020 17:12:42
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 00020000000032363435323751110037001a
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
MPU6000 on SPI bus 1 at 56 (1000 KHz)
BMI088_ACCEL on SPI bus 5 at 106 (10000 KHz)
BMI088_GYRO on SPI bus 5 at 102 (10000 KHz)
DPS310_SPI on SPI bus 2 at 104 (10000 KHz)
INFO  [ist8310] no device on bus 1
WARN  [hmc5883] no device on bus 1 (type: 2)
ERROR [hmc5883] bus 2 not started
ERROR [hmc5883] bus 2 not started
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS2
Starting MAVLink on /dev/ttyS0
nsh: mavlink: too many arguments
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS0 @ 57600B
ERROR [mavlink] offboard mission init failed (-1)
ERROR [param] Parameter DSHOT_CONFIG not found
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter SENS_EN_PX4FLOW not found
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> WARN  [PreFlightCheck] Preflight Fail: Compass Sensor #0 missing

The compass is the AK09916 inside the ICM-20948

davids5, 2020-03-11:
@pkocmoud 

What else needs to be done here?

pkocmoud, 2020-03-11:
Is there support for the AK09916 compass inside the ICM-20948

davids5, 2020-03-11:
@pkocmoud - I see the the AK09916 referenced in code. @dagar what is missing?

pkocmoud, 2020-03-11:
@davids5 does it matter that the wrong MCU model number is printed during bootup?

pkocmoud, 2020-03-11:
It seems counter productive that we lump all Invensense/TDK devices as MPU6000, that chip is obsolete and can cause confusion to if the proper device is even detected.

pkocmoud, 2020-03-12:
External I2C devices are not detected. Do you think this just a board config issue?

pkocmoud, 2020-03-30:
@dagar @davids5 - is there anything I can do to help push this along? Almost daily users are asking to use this board with PX4.

pkocmoud, 2020-04-02:
An update - I2C is fixed, just waiting on the ICM20948 driver to be completed and tested.

dagar, 2020-04-02:
New icm20948 driver support. https://github.com/PX4/Firmware/pull/14567

The IMU side was working, but broken at the moment as I work on enabling the AK09916 properly.

Ecuashungo, 2020-08-13:
Does this have something to do with #15541?

LorenzMeier, 2021-01-02:
@pkocmoud Looks like we have all the prerequisites managed (driver support there, etc). Could you test the current configuration so we have a baseline?

dagar, 2021-01-02:
I believe everything here has been resolved.

closed 2021-01-02 by dagar