PX4/PX4-Autopilot #14361: Latest Stable 1.10.1 can to bootup on 3DR Pixhawk Mini

bys1123, 2020-03-11:
Tested with Holybro the latest stable 1.10.1 can flash on 3DR Pixhawk Mini, but can't bootup
[image]

julianoes, 2020-03-26:
@bys1123 any way you could get the console boot log? Or `dmesg` output?

bys1123, 2020-03-27:
[image]
@julianoes

julianoes, 2020-03-30:
Seems like the IO fails to connect. Can you try to boot it up while keeping the safety button pressed to force an IO update?

bys1123, 2020-04-01:
@julianoes It works, is this normal? or maybe need some change in code?

julianoes, 2020-04-01:
@bys1123 so now it works? No it's not normal, it shouldn't happen :confused:.

bys1123, 2020-04-01:
@julianoes Yes, keep press safety button on bootup will work. if not , autopilot will not bootup after firmware flash.

julianoes, 2020-04-01:
Ok, so even after it has once flashed correctly. Odd. Is that only the case with this hardware or can you reproduce with other hardware as well?

I can try to reproduce with my old Pixhawk here, not sure if that would be the same.

bys1123, 2020-04-01:
reproduced on lots of fmuv3

bys1123, 2020-04-01:
Also get few report say 1.10.1 can running on their 3DR Pixhawk mini

julianoes, 2020-04-06:
I've tried stable on a Pixfalcon which is still fmu-v2 and `px4io status` worked for me once I had a airframe selected (quadcopter). `px4io` was not started without a aiframe selected. I'm not sure if that's supposed to be that way or if it's a bug.

closed 2020-07-05 by bys1123