PX4/PX4-Autopilot #14368: distance_sensor tfmini can't work

jinchengde, 2020-03-12:
**Describe the bug**
FW version 1.10.1 work good
in master , tf mini can't get the data from MAVLINK, no distance sensor
[image]
and listener the distance_sensor 
[image]
the current_distance in topic can't change any more if you move the distance sensor

473867143, 2020-03-20:
My TFMiNi can work well.
To reproduce:
1. take a newly flashed Pixhawk4 board with PX4 1.10.0 stable
2. Plug TFMiNi LIDAR to Serial 4 port (UART &I2CB)
3. Connect the board to USB
4. Launch QGC, set SENS_TFMINI_CFG to 'TELEM/SERIAL 4 ', and set EKF2_HGT_MODE to 'Range sensor',finaly reboot the board.
5. Open analyze widget, the distance sensor variable will be present.

[image]

hopefully it helped you!

jinchengde, 2020-03-21:
> My TFMiNi can work well.
> To reproduce:
>
> 1. take a newly flashed Pixhawk4 board with PX4 1.10.0 stable
> 2. Plug TFMiNi LIDAR to Serial 4 port (UART &I2CB)
> 3. Connect the board to USB
> 4. Launch QGC, [...]

yes, the stable version is OK(include v1.10.1), but the develop versin in master can't work.

bkueng, 2020-03-23:
This should be fixed in https://github.com/PX4/Firmware/pull/14423. @jinchengde can you test master again?

jinchengde, 2020-03-24:
@bkueng it works

bkueng, 2020-03-24:
Thanks for testing. Closing this.

closed 2020-03-24 by bkueng