PX4/PX4-Autopilot #14369: feature request : Adding roll , pitch actuator to mavlink massages

moreba1, 2020-03-12:
Hi

adding control[0] , control[1] for roll and pitch of fixedwing and multirotor
This will help we that display aileron and elevator percent in QGC and monitor that dont to be saturated during flight . also we can use this when we reply log for flight analysis in QGC . 

[image]

julianoes, 2020-03-26:
Is the data already in the ulog and just lacking in flight review or is it not logged?

moreba1, 2020-03-26:
the data already logged in ulog but is not in mavlink massages and tlog for online display.

julianoes, 2020-03-26:
Got it. For the mavlink messages you could suggest a change for `mavlink_messages.cpp` to add it. For flight review you could make a PR here: https://github.com/PX4/flight_review

dayjaby, 2020-03-30:
There are already the correct mavlink streams available.

You can enable them, e.g. by adding this line to etc/extras.txt (see https://dev.px4.io/v1.10/en/concept/system_startup.html)

```
mavlink stream -d /dev/ttyS0 -s ACTUATOR_CONTROL_TARGET0 -r 50
mavlink stream -d /dev/ttyS0 -s ACTUATOR_CONTROL_TARGET1 -r 50
```
ACTUATOR_CONTROL_TARGET1 is not built by default. Make sure to duplicate the line https://github.com/PX4/Firmware/blob/536cd6cb1af68e3b48f310a0b42b77faf974e4e1/src/modules/mavlink/mavlink_messages.cpp#L5200 and replace the 0 with a 1.

closed 2020-03-30 by dagar

moreba1, 2020-08-04:
> There are already the correct mavlink streams available.
>
> You can enable them, e.g. by adding this line to etc/extras.txt (see https://dev.px4.io/v1.10/en/concept/system_startup.html)
>
> ```
> mavlink [...]

Hi 

how where we can add the first line to code instead of sd card?