PX4/PX4-Autopilot #14381: px4_fmu-v2_default VTOL upgrade handling

dagar, 2020-03-12:
Due to flash constraints we had to comment out the VTOL controller from px4_fmu-v2_default builds (https://github.com/PX4/Firmware/pull/14342). We need to add logic at some level to notify any potential users.

hamishwillee, 2020-03-12:
We should also add docs update to this effect. 
- Will VTOL airframes still show up for FMUv2 boards? If not, we could add a note to https://docs.px4.io/master/en/config/airframe.html
- What _exactly_ should users do if they want to use VTOL on an FMUv2 board?

We might:
- Update the first introduction to FC boards with this info https://docs.px4.io/master/en/getting_started/flight_controller_selection.html
- Explain the problem near "recommended boards": https://docs.px4.io/master/en/flight_controller/pixhawk_series.html#recommended

hamishwillee, 2020-03-19:
@mrpollo The release notes indicated this would be going into 1.11 - is that the case? If so, it would be nice if you someone (@dagar ?) could answer the questions in https://github.com/PX4/Firmware/issues/14381#issuecomment-598427618 - 

Bit pointless for me to spend the time building and testing this when someone in the conversation could spend 45 seconds telling me.

mrpollo, 2020-04-06:
@hamishwillee agreed, this needs to be resolved, I will chase down the answers and post-back here

closed 2020-04-15 by julianoes