PX4/PX4-Autopilot #14385: Proposal - use enum on uorb

BazookaJoe1900, 2020-03-13:
**Describe problem solved by the proposed feature**
uorb can't use enum as type for its fields. by using, some issues that can be fixed:
* tighter type protections
* less duplication of similar types between uorb messages, Mavlink messages, and c Types
* less needs of conversions

**Describe your preferred solution**
It can be solved by few ways:
**1) declare the enum on the urob.msg file, something like that:**
```
enum NAVIGATION_STATE uint8 # Navigation state, i.e. "what should vehicle do".
NAVIGATION_STATE::MANUAL = 0		# Manual mode
NAVIGATION_STATE::ALTCTL = 1		# Altitude control mode
NAVIGATION_STATE::POSCTL = 2		# Position control mode
NAVIGATION_STATE::AUTO_MISSION = 3	# Auto mission mode
...
end enum NAVIGATION_STATE
```
that will create enum (or enum class) during the message generation. this enum can be used on the code this way:
on the cpp files code instead of:
`uint8_t nav_state = vehicle_status_s::NAVIGATION_STATE_AUTO_MISSION;
`
do:
`vehicle_status_s::NAVIGATION_STATE nav_state = vehicle_status_s::NAVIGATION_STATE::AUTO_MISSION;`

pros: 
 - uorb keeps self sustained system, 
 - the enum can be generated to different languages and decoders. 

cons: 
 - harder to share enums between uorbs (that can be solved by declaring the enums outside of the msg file but in some kind of msg_h file)

**2) enable external enum as type in uOrb, something like that:**

declare NAVIGATION_STATE in external h file.
on .msg file, instead of:
`uint8 nav_state		# set navigation state machine to specified value
`
do:
`NAVIGATION_STATE :uint8 nav_state	# set navigation state machine to specified value
`

pros: 
 - can reuse enum between mavlink/c functions/uorb without the need to generate the uorb

cons: 
 - uorb isn't self sustained system, the enum can't be generated to different languages, 
 -

dagar, 2020-03-13:
I'm all for this if we can find a way to transition.

BazookaJoe1900, 2020-03-13:
@dagar I can start to do proof of concept, which direction you prefer? 1st? 2nd? other notes?

dagar, 2020-03-13:
Option 1 if possible, but that gets into message generation compatibility issues (ROS2, etc). 
https://github.com/PX4/px4_msgs
https://github.com/PX4/px4_ros_com

@TSC21 any idea if we could start doing proper enums in messages?

TSC21, 2020-03-13:
If you are willing to update the IDL template (https://github.com/PX4/Firmware/blob/master/msg/templates/urtps/msg.idl.em) to properly handle the enumeration parsing, then I am fine with any option :)

TSC21, 2020-03-13:
Ah that's one way of the pipeline. Because the other way around would mean that whatever is parsed on the RTPS bridge needs to be converted back to the enumeration structure that is being proposed. I don't see an easy way of doing that. uORB as is is quite well suited to be parsed to ROS/RTPS msgs and back.

BazookaJoe1900, 2020-03-13:
@dagar what did you mean by etc? what other generator there are except for ROS2?
using enum while writing and reading uorb doesn't necessarily means that all the other writing and readers need that enum. 
It can be build step by step (I think...)

dagar, 2020-03-13:
> @dagar what did you mean by etc? what other generator there are except for ROS2?
> using enum while writing and reading uorb on the C++ doesn't necessarily means that all the other writer and readers need that enum.
> It can be build step by step (I think...)

I mean ROS2 overall, but it's all the different pieces that would need to properly support the change. Things like genmsg_cpp, https://github.com/PX4/px4_msgs, ROS2 actually supporting enums.

BazookaJoe1900, 2020-03-20:
First draft for the proposal:
https://github.com/BazookaJoe1900/genmsg/tree/enums
https://github.com/BazookaJoe1900/Firmware/tree/pr-uob_enums

What was implemented there is the ability to define enums inside the uorb message file in this syntax:
https://github.com/BazookaJoe1900/Firmware/blob/4bc496d59e223ad4b4cd8b9d84973a8b8e7d659a/msg/vehicle_command.msg#L83:L90

currently I implemented that on vehicle_command.msg.

The current main design question is where to define the enums.
In this implementation I just kept it on the uorb. but it instantly raised issue that both vehicle_command_ack and vehicle_command has same enum to use.
So you will see a lot of 'not really needed' casts.
my thought is to define new x.mgs_enum file, where enums can be defined, and to be cast to different languages syntax.

I have added the generated h and cpp files to allow code review without need to use my genmsg package

TSC21, 2020-03-20:
> First draft for the proposal:
> https://github.com/BazookaJoe1900/genmsg/tree/enums
> https://github.com/BazookaJoe1900/Firmware/tree/pr-uob_enums
>
> What was implemented there is the ability to define [...]

I warn again that this change will not be properly supported in `px4_msgs` and on the microRTPS/ROS2 side, unless the proper handlers and parsing is done.

BazookaJoe1900, 2020-03-20:
@TSC21, its understood that this idea need to fit to all of the dependencies of uorb, including the ROS2 and RTPS mechanism. after (and if...) the concept set the enums will be decided, we can see how it can be tail to ROS2

TSC21, 2020-03-20:
> @TSC21, its understood that this idea need to fit to all of the dependencies of uorb, including the ROS2 and RTPS mechanism. after (and if...) the concept set the enums will be decided, we can see how it can be tail to ROS2

If the decision is taken over an implementation of the enumerations, it needs to respect the overall architecture where uORB is used and where it plugs to. It will never be one thing and then the other because the RTPS bridge is already part of the architecture.

BazookaJoe1900, 2021-07-04:
@TSC21
seems that there is no reason to keep this issue open. closing

closed 2021-07-04 by BazookaJoe1900