PX4/PX4-Autopilot #14387: PX4 micrortps_client works only at /dev/ttyACM0

nicht762, 2020-03-13:
Hello guys! Sorry fo my English

**Describe the bug**
When I'm starting micrortps_client at /dev/ttyACM0 port on my Pixhawk 4 mini connected via microUSB-USB cable to my PC it works good - I can see information in ROS2. But when I'm starting micrortps_client at /dev/ttyS1 port (TELEM 1) - I have only bad CRC errors in micrortps_agent and can not see any information in ROS2.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the pixhawk with v1.9.0 (px4_fmu-v5_rtps)
2. Connect to Pixhawk debug port
3. Run micrortps_agent -d /dev/ttyS1 -b 115200
4. Connect Pixhawk TELEM 1 port to PC via USB-UART device (CH340G, FT232, etc)
5. Run micrortps_agent on PC
6. Run px4_ros_com sensor_combined_listener for example
7. See no msgs

I will provide any information if necessary.
Thanks!

TSC21, 2020-03-24:
What else do you have running in `/dev/ttyS1`? Do you have a Mavlink stream there?

Also, 1.9.0 is lacking a lot of features and bug fixes compared to master.

nicht762, 2020-03-28:
No, nothing else running in /dev/ttyS1. Mavlink is stopped anywhere.
I tried to compile later versions, but they did not work on my Pixhawk.
Maybe I'm doing something wrong.