PX4/PX4-Autopilot #14390: MS5803 driver for water pressure/depth measurement.

kefey6ecb, 2020-03-14:
Hello. I'm new to pixhawk development. Now I'm working on a project which is quick similar to the unmanned underwater vehicle called 'HippoCampus'. I would like to utilize MS5803-14bar as a pressure sensor to measure the water pressure, and convert the pressure reading to the depth of the vehicle. I found the ms5803 driver from repo of HippoCampus, which is this [https://github.com/EugenSol/HippoC](url). 

I have read the instruction in this link, [https://dev.px4.io/master/en/sensor_bus/i2c.html](url)

Are the driver able to work by copying the ms5803 driver folder into src/driver/barometer and add the driver to the cmake config file that corresponds to the target I want to build?

If not, which part of the code I need to modify?

bkueng, 2020-03-16:
Hi. Yes that should work. You'll also need to add it to the startup script: https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board_sensors.

kefey6ecb, 2020-03-16:
> Hi. Yes that should work. You'll also need to add it to the startup script: https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board_sensors.

Thank you @bkueng ! I connect MS5803 through I2C Bus.  In rc.board_sensors file, I found the codes below
`# External I2C bus
hmc5883 -C -T -X start
lis3mdl -X start
ist8310 -C start
qmc5883 -X start`

What are -C, -T and -X indicated? How can I write the code to initialize ms5803?

kefey6ecb, 2020-03-16:
Dear @DanielDuecker, What do you think regarding to this topic?

DanielDuecker, 2020-03-17:
Hi @kefey6ecb ,
those  are parameters (e.g. mounting rotation etc)  when starting the sensor
Check this out, as an example for a magnetometer https://github.com/PX4/Firmware/blob/9275937b4403b5614f8730db9f916a06355cf563/src/drivers/magnetometer/hmc5883/hmc5883_main.cpp#L318

I don't think you need any of those for the moment. However, Eugen's implementation is quite old, we are currently updating/extending px4's underwater robot/HippoCampus support.
Please feel free to contribute