PX4/PX4-Autopilot #14400: cross compilation for navio2 no longer works

cyrili, 2020-03-16:
**Describe the bug**
cross-compiling px4 for navio2 on my ubuntu 18lts pc fails. this issue only appears with the current master branch (v1.10.1 works fine).

"make emlid_navio2_cross" and "make emlid_navio2_native" dont exist anymore apparently, and there is now only "make emlid_navio2" which does not work for cross compilation but does work for native compilation. ([this ](https://github.com/PX4/Firmware/commit/7241eebd80086c136fc2026bacf0337f99f92176)is the relevant commit?)

**To Reproduce**
Steps to reproduce the behavior:
1. install necessary toolchains as per [tutorial](https://dev.px4.io/v1.9.0/en/setup/dev_env_linux_ubuntu.html#raspberry-pi-hardware)
2. git clone https://github.com/PX4/Firmware.git
3. cd Firmware
4. make emlid_navio2 

**Expected behavior**
px4 is cross-compiled for rpi+navio2 successfully

**Log Files and Screenshots**
<details>
   <summary>"make emlid_navio2" output</summary>

```
-- PX4 version: v1.11.0-beta1-787-g1a9452e411
-- PX4 config file: /home/me/test/Firmware/boards/emlid/navio2/default.cmake
-- PX4 config: emlid_navio2_default
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 4.8.3
-- The C compiler identification is GNU 4.8.3
-- The ASM compiler identification is GNU
-- Found assembler: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-239-gf1aa53d
-- Configuring done
-- Generating done
-- Build files have been written to: /home/me/test/Firmware/build/emlid_navio2_default
[7/815] git submodule src/drivers/gps/devices
[10/815] Building CXX object boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/i2c.cpp.o
FAILED: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++  -DCONFIG_ARCH_BOARD_EMLID_NAVIO2 -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/emlid/navio2/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../boards/emlid/navio2/include -I../../platforms/posix/include -Iexternal/Install/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -std=gnu++1y -MD -MT boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/i2c.cpp.o -MF boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/i2c.cpp.o.d -o boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/i2c.cpp.o -c ../../boards/emlid/navio2/src/i2c.cpp
In file included from ../../platforms/posix/src/px4/generic/generic/include/px4_arch/i2c_hw_description.h:36:0,
                 from ../../boards/emlid/navio2/src/i2c.cpp:34:
../../platforms/common/include/px4_platform_common/i2c.h: In constructor ‘I2CBusIterator::I2CBusIterator(I2CBusIterator::FilterType, int)’:
../../platforms/common/include/px4_platform_common/i2c.h:86:3: error: declaration of ‘bus’ shadows a member of 'this' [-Werror=shadow]
   : _filter(filter), _bus(bus) {}
   ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[10/815] Building CXX object boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/spi.cpp.o
FAILED: /home/me/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-g++  -DCONFIG_ARCH_BOARD_EMLID_NAVIO2 -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/emlid/navio2/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../boards/emlid/navio2/include -I../../platforms/posix/include -Iexternal/Install/include -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -std=gnu++1y -MD -MT boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/spi.cpp.o -MF boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/spi.cpp.o.d -o boards/emlid/navio2/src/CMakeFiles/drivers_board.dir/spi.cpp.o -c ../../boards/emlid/navio2/src/spi.cpp
In file included from ../../platforms/posix/src/px4/generic/generic/include/px4_arch/spi_hw_description.h:36:0,
                 from ../../boards/emlid/navio2/src/spi.cpp:34:
../../platforms/common/include/px4_platform_common/spi.h: In constructor ‘SPIBusIterator::SPIBusIterator(SPIBusIterator::FilterType, uint16_t, int)’:
../../platforms/common/include/px4_platform_common/spi.h:145:3: error: declaration of ‘bus’ shadows a member of 'this' [-Werror=shadow]
   : _filter(filter), _devid_driver_index(devid_driver_index),
   ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[10/815] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'emlid_navio2' failed
make: *** [emlid_navio2] Error 1

```

</details>

SalimTerryLi, 2020-03-16:
Oh, that isn't what you think of. Just because the toolchain provided by RPi official repo is out-of-date.
Documents have been updated on the master branch. Please refer to [here](http://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html)

navioglider, 2020-03-16:
oh crap, now i see that i was on the outdated wiki. damn google sent me to the wrong page.. thanks a lot

cyrili, 2020-03-16:
oh crap.. i didn't see that i was on the wiki for the older version. damn google for sending me there.. thanks a lot, now it compiles!

SalimTerryLi, 2020-03-16:
> oh crap.. i didn't see that i was on the wiki for the older version. damn google for sending me there.. thanks a lot, now it compiles!

Congratulations. And don't forget to close this issue :)

closed 2020-03-16 by LorenzMeier

brianthelion, 2020-03-16:
@SalimTerryLi Please note however that the documentation has not been updated for the [native build](https://dev.px4.io/master/en/setup/building_px4.html#native-build).

```bash
$ git log | head -7
commit 9275937b4403b5614f8730db9f916a06355cf563
Author: Lorenz Meier <lorenz@px4.io>
Date:   Mon Mar 16 01:27:45 2020 +0100

    Commander: Exclude shell commands for constrained flash
    These are convenience commands for developers and are not required for flash constrained targets.

$ make emlid_navio2_native
Makefile:473: *** "Make target emlid_navio2_native not found. It either does not exist or emlid_navio2_native cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
```