PX4/PX4-Autopilot #14402: [WIP] v1.11.0 Release Notes

mrpollo, 2020-03-16:
# v1.11 Release Notes

Following the great stability improvements on v1.10, the PX4 team is preparing for a new release which will see lots of quality of life improvements, as well as internal robustness in the system architecture to allow for easier board configuration.

## Teamwork

[image]

## Community we need your help

Please review the changelog and help us identify anything that should be part of the high-level release notes.

---

# NOTE: This is still considered Work In Progress

## Release Notes
### Highlights
* Major sensor calibration improvements ([Gyro](https://github.com/PX4/Firmware/pull/13632), [Mag](https://github.com/PX4/Firmware/pull/13623), [Level](https://github.com/PX4/Firmware/pull/13666))
* Enable RC-Override in Take-off (PR  #14022)
* Pre-flight check: Disable mag field strengh check via parameter (COM_ARM_MAG_STR) (PR #13766)
### Main Features and Improvements
### Migration Guide
* **DEPRECATED:** VTOL is no longer supported in FMUv2 hardware
### Multicopter
* New hover thrust estimator (deactivated by default)
* New Notch filter (PR #13549)
* Acceleration feedforward (PR #14212)
### VTOL
* Tiltrotor: Use tilt to accelerate the vehicle forward while in hover (PR #13779)
* Weather vane improvements (PR #13782)
### Fixed Wing
### Rover
* Major improvements to offboard control
* Enable offboard attitude setpoints (PR #13323)
* Enable offboard velocity setpoints (PR #13225)
* Enable ActuatorControl setpoints in offboard mode (PR #13314)
### Underwater Vehicles
* New mixers for underwater vehicles Generic & HippoCampus (PR #14079)
* New UUV default specific parameters and autostarting (PR #14079)
* New Mixer for UUVs with X-shaped thruster configuration (PR #14079)
* New `uvv_att_control` controller for UUV attitude control (PR #14079)
### PX4/ECL - Estimators improvement
* Emergency Yaw recovery using EKF-GPS estimator (PR PX4/ecl#766)
### PX4/Drivers
* Enable SPI DMA adds centralized DMA map (PR #14207)
* Move primary IMU to new InvenSense driver (PR #14356)
  * FMUv5, FMUv5x
  * ModalAI_fv-v1
### Onboard computer / middleware improvements
* Adds ROS 2, Dashing & Eloquent support to microRTPS bridge (PR #13907)
* Adds time sync to microRTPS bridge (PR #14297)
### Computer Vision & Autonomy
### PX4/Avoidance
### Simulation
### Logging improvements
### Failsafe / commander improvements

### OS/NuttX
The release includes a change from [NuttX-7.29 ](https://bitbucket.org/nuttx/nuttx/src/101f3bddecb70cfb8dc78c5b4fad98f3480ae399/ReleaseNotes#lines-21789) to [NuttX-8.2](https://bitbucket.org/nuttx/nuttx/src/efff9b9438e69ea6c70e879b7ee4d49bff1b0988/ReleaseNotes#lines-25651)
See [NuttX release notes](https://bitbucket.org/nuttx/nuttx/src/efff9b9438e69ea6c70e879b7ee4d49bff1b0988/ReleaseNotes#lines-25651) for the details of changes in NuttX-8.2 details.

Included as backports are:
* STM32[F4 F7 H7]
  * spi_exchange (no DMA) with for STM32xx_SPI_DMATHRESHOLD
* STM32H7
  * fix stm32h7x3xx_dmamux.h: add missing underscore to defines
* STM32F7 and STM32H7 SDMMC internal pull up usage fixed 
  * Add check_format tooling 
  * nxstyle - output compiler like error format
  * stm32f7:Add Serial Tx DMA 
  * stm32f7:ethernet: Add some delays so that ifup() does not
* NXP IMX-RT, S32K
  * S32K add support for Nxp drone boards (#224) 
  * imxrt: Adds the ability to run from OCRAM (#407) 
  * imxrt: Add USB Device support for i.MX RT (#408) 
  * imxrt: added missing i2c prescale mask
  * imxrt: interrupt serial storm, add DTCM and set up I and
  * imxrt: Allow clock setting for SPI and I2C from board.h.
  * imxrt:lpi2c Fix interrupt storm on failed write. 
  * imxrt:lpi2c ensure that on an error status reflects it. 
  * imxrt:lpi2c imxrt_lpi2c_reset uses GPIO with SION 
  * imxrt:gpio Support readback on OUT GPIO 
  * imxrt1020-evk: Enable the GPIO based CD.
  * imxrt:lpspi: Fixed race on register setting. 
  * imxrt:lpspi: Remove hack setting LPSPI1 daisy irrespective 
  * imxrt:lpi2c.c:Added configurations to fine tune LPI2C Time 
  * IMXRT106x USDHC: Support regular GPIO for CD and inversion. 
  * IMXRT106x:pinout add ALT 8 GPIO_GPT2_COMPARE3 & fix GPIO_G 
  * IMXRT106x:pinout add ALT 8 GPIO_GPT1_CAPTURE[1|2] 
  * Kinetis renamed TJA1100 to TJA110X registers

### Updated Supported Hardware
* Holybro Durandal
* NXP IMX-RT, S32K

## v.1.11.0 Changelog
Link upcoming

bresch, 2020-03-17:
I'm not exactly sure what should go under "Highlights". Shouldn't be the "overall best improvements and new features" ?
Enable RC-Override in Take-off (PR #14022) and Pre-flight check: Disable mag field strengh check via parameter (COM_ARM_MAG_STR) (PR #13766) are minor modifications and shouldn't be here.

Other comments:
> Emergency Yaw recovery using EKF-GPS estimator

EKF-**GSF**

> Navigation improvements

**Navigator** improvements

> New hover thrust estimator (

https://github.com/PX4/Firmware/pull/13981

jinger26, 2020-03-17:
Is "Computer Vision & Autonomy" redundant to "PX4/Avoidance"?

mrpollo, 2020-03-17:
Thanks for the edits @bresch, please let me know if the changes work for you.

@jinger26 Yes. A remnant of the last release notes, not sure I remember what was going on there right now. Take a look `->` [PX4 v1.10.0](https://github.com/PX4/Firmware/releases/tag/v1.10.0). It's gone now, thank you.

mrpollo, 2020-03-25:
add to notes please 

https://github.com/PX4/Firmware/pull/14473

mrpollo, 2020-03-25:
add to notes please

https://github.com/PX4/Firmware/pull/14405

mrpollo, 2020-04-08:
update notes 

https://github.com/PX4/Firmware/pull/14473

ThomasRigi, 2020-04-15:
Is dual battery support in v1.11? https://github.com/PX4/Firmware/pull/12551 and https://github.com/PX4/Firmware/pull/12034 

They're not in the list (yet)

hamishwillee, 2020-07-20:
@mrpollo FMI is this ever going to release?

jinger26, 2020-08-12:
Add to supported hardware:
Holybro PIX32V5
HEX Yellow/Orange?
CUAV X7, X7Pro, Nora?

verify with @dagar

mrpollo, 2020-09-07:
Thanks, everyone, the release notes are finally complete.

closed 2020-09-07 by mrpollo