PX4/PX4-Autopilot #14422: Logs are unable to be used by the ekf2 replay facility

priseborough, 2020-03-19:
To replicate the fault:

1) Use 'make px4_sitl jmavsim' to start the multicopter flight simulator
2) Command a takeoff using 'commander takeoff' from the console
3) Wait for the comms failsafe to land the vehicle, disarm and close the logfile.
4) Follow the instructions here https://dev.px4.io/v1.9.0/en/debug/system_wide_replay.html#ekf2-replay to run the replay using the file generated by the SITL simulation.

The replay will not publish any topics and generates the following error message:

pxh> ERROR [replay] unknown log definition type 76, size 45 (offset 42754)

The commit responsible is d4c6982dbd306ea94581c0348b02842215498145

A SITL log file generated using an earlier commit can be replayed using a later version of the SW, but not vice-versa indicating this could be a uORB of logger issue.

priseborough, 2020-03-19:
This is a SITL log from the first commit that is unable to be used for a ekf2 replay: https://logs.px4.io/plot_app?log=31d6f586-0a03-4ed0-aff0-5ec396c8315d

This is a SITL log from the first commit that is able to be used for a ekf2 replay:
https://logs.px4.io/plot_app?log=aa2f145e-3b80-466c-b6a8-810fdf8451ae

These will be useful to see if there are any obvious differences

priseborough, 2020-03-19:
There is nothing between them wrt sensor timestamp and timestamp_relative messages and the same sensor data fields are there for both as far as I can see.

dagar, 2020-03-19:
Possible culprits.
 - uORB changes https://github.com/PX4/Firmware/pull/14106
 - `vehicle_air_data` additional fields added https://github.com/PX4/Firmware/pull/14096

bkueng, 2020-03-20:
@priseborough I found the problem and fixed it in https://github.com/PX4/Firmware/pull/14435. Can you quickly test it as well?

closed 2020-03-20 by bkueng

priseborough, 2020-03-24:
@bkueng Sorry, I was off on another project and missed this. Yes it does fix the issue in my testing. Thanks for fixing it.