PX4/PX4-Autopilot #14430: make px4_sitl gazebo --> c++: internal compiler error: Killed (program cc1plus)

MuhammadBilal1, 2020-03-19:
**Describe the bug**
I am trying to build the code of PX4 that I got from [here](https://github.com/PX4/Firmware.git). After running the "make px4_sitl gazebo" command in terminal, I am getting the below issue of "internal compiler error:"

**OS:** Ubuntu 18.04.04 x64 inside a VMWare environment.

My main objective is to connect Gazebo with QGroundControl, but getting hurdles while setting up the things.

Following the instructions given in this [YouTube video](https://www.youtube.com/watch?v=-zEddRFbMvQ)

### **Here is the terminal output:** (I made some formatting of log)

bash: /home/myuser/catkin_ws/devel/setup.bash: No such file or directory
myuser@ubuntu:~/Downloads$ cd
myuser@ubuntu:~$ cd src/Firmware/
myuser@ubuntu:~/src/Firmware$ sudo make px4_sitl gazebo
[sudo] password for myuser: 
[0/4] Performing build step for 'sitl_gazebo'
[17/43] Building CXX object CMakeFiles...dir/src/gazebo_mavlink_interface.cpp.o
FAILED: CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_mavlink_interface_EXPORTS -isystem 

/usr/include/gazebo-9 -isystem 
/usr/include/bullet -isystem 
/usr/include/simbody -isystem 
/usr/include/sdformat-6.0 -isystem 
/usr/include/ignition/math4 -isystem 
/usr/include/OGRE -isystem 
/usr/include/OGRE/Terrain -isystem 
/usr/include/OGRE/Paging -isystem 
/usr/include/ignition/transport4 -isystem 
/usr/include/ignition/msgs1 -isystem 
/usr/include/ignition/common1 -isystem 
/usr/include/ignition/fuel_tools1 -isystem 
/usr/include/x86_64-linux-gnu/qt5 -isystem 
/usr/include/x86_64-linux-gnu/qt5/QtCore -isystem 
/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ 
-I/home/myuser/src/Firmware/Tools/sitl_gazebo/include 
-I. -I/usr/include/eigen3 
-I/usr/include/eigen3/eigen3 
-I/usr/include/gazebo-9/gazebo/msgs 
-I/home/myuser/src/Firmware/mavlink/include 
-I/usr/include/Paging -isystem 
/usr/include/opencv 
-I/home/myuser/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include 
-I/usr/include/gstreamer-1.0 
-I/usr/include/glib-2.0 
-I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem 
/usr/include/uuid -isystem 
/usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -Wno-address-of-packed-member -fPIC   
-I/usr/include/uuid 
-I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT 

CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o -MF CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o.d -o CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o -c /home/myuser/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[30/43] Building CXX object CMakeFiles...n.dir/src/gazebo_catapult_plugin.cpp.o
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/myuser/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
myuser@ubuntu:~/src/Firmware$ 123

julianoes, 2020-03-26:
This usually happens if you run out of RAM. Can you try to raise the RAM available for the VM and try again?

MuhammadBilal1, 2020-03-26:
@julianoes : Thanks for responding. I just tried the solution you told, but this didn't worked. I am still getting the issue. I tried to run 2 simulators (jmavsim & gazebo), but both are not running/connecting with my QGroundControl. 

I increased the RAM of my VM to 9.9GB, and its already using the 16 Processor Cores.

Detailed log is given below for both.
[gazebo_log.txt](https://github.com/PX4/Firmware/files/4386943/gazebo_log.txt)
[jmavsim_log.txt](https://github.com/PX4/Firmware/files/4386944/jmavsim_log.txt)

[image]

julianoes, 2020-03-26:
For Gazebo: There is an undefined symbol, so it's a linking problem.
Make sure the gazebo dependencies are installed correctly and up-to-date. Maybe run the `ubuntu.sh` script again to install all dependencies.

For jMAVsim: https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting

julianoes, 2020-05-25:
@MuhammadBilal1 It looks like RAM is still too low because you have so many cores. I suggest to enable less CPU cores for VirtualBox and that way preventing it from doing too much in parallel while trying to compile the gazebo plugins.

julianoes, 2020-06-04:
With #14971 you can now build with fewer cores in order to avoid running out of RAM:

```
make px4_sitl gazebo -2
```

closed 2020-06-04 by julianoes

MuhammadBilal1, 2020-06-04:
> lel
@julianoes 

Thank you. Your suggestions worked. This ticket can be closed now.
I decreased my cores to 2 and processors to 2 and it build successfully.
**These are the specifications of my VM that worked.**
Number of processors : 2
Number of cores per processor: 2
Total processor cores: 4
RAM: 8GB
OS: Ubuntu 18.04.04

[image]

julianoes, 2020-06-04:
Cool, thanks for the feedback @MuhammadBilal1!