PX4/PX4-Autopilot #14436: Multi-Vehicle Simulation with Gazebo using SDF, can only control first drone.

bavokn, 2020-03-20:
Hi, i'm trying to spawn multiple PX4's with a depth camera and a lidar but i can only control the first drone spawned. I use some custom scripts to setup and start gazebo in a docker container and generate a .launch file. 

first I start the simulation in a container using a script 

```
#!/bin/bash

# xhost +local:docker

# --device=/dev/video0:/dev/video0
# For non root usage:
# RUN sudo usermod -a -G video developer

TITLE='echo -ne "\033]0;Simulation\007"'
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
vendor=$(glxinfo | grep vendor | grep OpenGL | awk '{ print $4 }')
$AMOUNT_UAVS

if [ $# -eq 0 ]; then
	echo "Didn't define amount of uavs: default 1"
    AMOUNT_UAVS="1"
else
	AMOUNT_UAVS=$1
fi

if [ "$vendor" == "NVIDIA" ]; then
	nvidia-docker run --privileged -it --rm \
		-e "DISPLAY" \
		-e "XAUTHORITY=$XAUTH" \
		-e "QT_X11_NO_MITSHM=1" \
		-e "TERM=xterm-256color" \
		-e "PROMPT_COMMAND=$TITLE" \
		-e "AMOUNT_UAVS=$AMOUNT_UAVS" \
		-e XDG_RUNTIME_DIR=/tmp/myxdg \
		-e "CONTAINER_NAME=simulation" \
		-e PULSE_SERVER=unix:"${XDG_RUNTIME_DIR}"/pulse/native \
		-v "$XAUTH:$XAUTH" \
		-v "/tmp/.X11-unix:/tmp/.X11-unix:rw" \
		-v "${XDG_RUNTIME_DIR}"/pulse/native:"${XDG_RUNTIME_DIR}"/pulse/native \
		-v "${DIR}"/models:/home/user/.gazebo/models/ \
		-v "${DIR}"/src/simulation:/home/user/catkin_ws/src/smartuav/ \
		--network=host \
		--runtime=nvidia \
		--device=/dev/snd \
		--device=/dev/fuse \
		--cap-add SYS_ADMIN \
		--name smartuav_simulation \
		--group-add "$(getent group audio | cut -d: -f3)" \
		smartuav_px4:latest \
		bash
else
	docker run --privileged -it --rm \
		-e "DISPLAY=$DISPLAY" \
		-e "TERM=xterm-256color" \
		-e "PROMPT_COMMAND=$TITLE" \
		-e "AMOUNT_UAVS=$AMOUNT_UAVS" \
		-e "CONTAINER_NAME=Simulation" \
		-v /tmp/.X11-unix:/tmp/.X11-unix \
		-v "${DIR}"/models:/home/user/.gazebo/models/ \
		-v "${DIR}"/src/simulation:/home/user/catkin_ws/src/smartuav/ \
		--network=host \
		--device /dev/fuse \
		--cap-add SYS_ADMIN \
		--device=/dev/dri:/dev/dri \
		--name smartuav_px4 \
		smartuav_px4:latest \
		bash
fi

source ~/.bashrc
[... 1411 lines cut ...]
log file: /home/user/.ros/log/b35c2186-6a88-11ea-8093-b06ebf5f23c9/iris_2-iris_2_spawn-15*.log
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_1/motor_speed/3, deleting message. This warning is printed only once.
INFO  [ecl/EKF] 5117000: reset position to GPS
INFO  [ecl/EKF] 5117000: reset velocity to GPS
INFO  [ecl/EKF] 5117000: commencing GPS fusion
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_1/motor_speed/0, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_1/motor_speed/1, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_2/motors, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_1/motors, deleting message. This warning is printed only once.
[ INFO] [1584698890.218655473, 132.168000000]: WP: mission received
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_1/motor_speed/2, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/iris_2/motor_speed/2, deleting message. This warning is printed only once.
```

as you can see, a few problems. What am i doing wrong ? The issue is pure in this line (generating the launch file) : 

```
<arg name="cmd" value="xmlstarlet ed -d '//plugin[@name=&quot;mavlink_interface&quot;]/14562' -s '//plugin[@name=&quot;mavlink_interface&quot;]' -t elem -n 14562 -v 14562 $(find smartuav)/models/iris_depth_camera_lidar/iris_depth_camera_lidar.sdf" />
```

I have tried switching the port numbers from udp to tcp but to no avail. The biggest problem is i dont know exaclty what some arguments do in this command.

**Drone :**
- PX4

julianoes, 2020-03-26:
@Jaeyoung-Lim any idea?

Jaeyoung-Lim, 2020-03-26:
@bavokn It seems like you are using the same sdf file for all vehicles. This will make the vehicles share the same tcp port for the HIL mavlink messages. 

You need to copy the sdf file and modify the `mavlink_tcp_port` as in https://github.com/PX4/sitl_gazebo/blob/master/models/rotors_description/urdf/component_snippets.xacro#L302 for each `iris.sdf`

bavokn, 2020-03-27:
If you look at the generation of the launch file you can see i set the mavlink_tcp_port for each drone diffrently :

```
<arg name=\"mavlink_tcp_port\" value=\"$(expr 4560 + $i)\" />\n\
```

which results in the launch file : 

UAV iris_0
```
<arg name="mavlink_tcp_port" value="4560" />
```

UAV iris_1
```
<arg name="mavlink_tcp_port" value="4561" />
```

UAV iris_2
```
<arg name="mavlink_tcp_port" value="4562" />
```

dayjaby, 2020-03-27:
This sets only the correct ports from the px4 side. Gazebo has to comply to those ports, but .sdf files cannot contain variables. So either you copy the .sdf file multiple times and change the port or you create a .xacro file which can contain variables.

@Jaeyoung-Lim did a PR recently for multi VTOL: https://github.com/PX4/sitl_gazebo/pull/398

bavokn, 2020-03-27:
Using xarco files worked , thank you !

closed 2020-03-27 by bavokn

petertheprocess, 2020-05-10:
@bacokn 
It seems that in your .sdf file, there is no tag( the tag is in the iris.sdf and you just include it, so xmlstarlet cannot find the tag):
`<plugin name='mavlink_interface' filename='libgazebo_mavlink_interface.so'>`
So you cannot use
`<arg name="cmd" value="xmlstarlet ed -d '//plugin[@name=&quot;mavlink_interface&quot;]/14562' -s '//plugin[@name=&quot;mavlink_interface&quot;]' -t elem -n 14562 -v 14562 $(find smartuav)/models/iris_depth_camera_lidar/iris_depth_camera_lidar.sdf" />
`to change the mavlink_tcp_port, though you set different
`<arg name="mavlink_tcp_port" value="4560" />` correctly

Besides `-n 14562 -v 14562` should be `-n mavlink_tcp_port -v 14562`