PX4/PX4-Autopilot #14439: Pixracer MAVLink over USB not working anymore

julianoes, 2020-03-20:
**Describe the bug**
As soon as a airframe is configured for Pixracer, MAVLink ovre USB does not seem to be working anymore. 

**To Reproduce**
1. Flash Pixracer
2. `param reset`
3. Choose airframe or do `param set SYS_AUTOSTART 4001`
4. Reboot and try to connect

**Log Files and Screenshots**

```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: 240aad1aef6ba2fe7c536f6577cb59a90a6047bc
FW version: 1.11.0 80 (17498240)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: befe53e7f1ed557b7239c9ed564acae7cc72a900
Build datetime: Mar 20 2020 14:41:47
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000100000000303833363335510d00300041
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
reset done, 50 ms
WARN  [lis3mdl] no device on bus 2
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [rm3100] no device on bus 2
MS5611_SPI on SPI bus 2 at 61 (20000 KHz)
ms5611 #0 on SPI bus 2 (devid=0x3d)
WARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 5 (11000 KHz)
WARN  [lis3mdl] no device on bus 3
WARN  [icm20602] unexpected WHO_AM_I 0x00
ERROR [icm20602] SPI::init failed
ERROR [icm20602] driver init failed
WARN  [icm20608g] unexpected WHO_AM_I 0x00
ERROR [icm20608g] SPI::init failed
ERROR [icm20608g] driver init failed
WARN  [mpu9250] probe failed! 0
WARN  [mpu9250] no device on bus 3
MPU9250 on SPI bus 1 at 36 (1000 KHz)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
nsh: mavlink: too many arguments
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] 9481081: EKF aligned, (baro height, IMU buf: 22, OBS buf: 14)

nsh>
nsh>
nsh>
nsh> INFO  [frsky_telemetry] Scanning timeout: exiting

nsh>
nsh>
nsh>
nsh> mavlink status

instance #0:
        mavlink chan: #0
        no radio status.
        flow control: OFF
        rates:
          tx: 0.000 kB/s
          txerr: 0.000 kB/s
          tx rate mult: 1.000
          tx rate max: 800000 B/s
          rx: 0.000 kB/s
        FTP enabled: YES, TX enabled: YES
        mode: Config
        MAVLink version: 1
        transport protocol: serial (/dev/ttyACM0 @57600)

instance #1:
        mavlink chan: #1
        type:           GENERIC LINK OR RADIO
        flow control: OFF
        rates:
          tx: 0.910 kB/s
          txerr: 0.000 kB/s
          tx rate mult: 1.000
          tx rate max: 20000 B/s
          rx: 0.000 kB/s
        FTP enabled: NO, TX enabled: YES
        mode: Normal
        MAVLink version: 1
        transport protocol: serial (/dev/ttyS0 @921600)

instance #2:
        mavlink chan: #2
        type:           GENERIC LINK OR RADIO
        flow control: OFF
        rates:
          tx: 1.004 kB/s
          txerr: 0.000 kB/s
          tx rate mult: 1.000
          tx rate max: 1200 B/s
          rx: 0.000 kB/s
        FTP enabled: YES, TX enabled: YES
        mode: Normal
        MAVLink version: 1
        transport protocol: serial (/dev/ttyS1 @57600)
nsh> 
```

LorenzMeier, 2020-03-20:
Can you run

```mavlink boot_complete```

And see if that fixes it? If it does, then that is NOT the problem (!), but merely a hint that the boot got aborted in the rcS and you need to find the location where the boot aborted.

julianoes, 2020-03-20:
On Pixhawk 4 it is erratic as well. Sometimes it connects, sometimes it does not. `mavlink boot_complete` does not fix it.

Probably it's yet another USB hub or USB cable problem.

julianoes, 2020-03-20:
Ok, it seems to be one faulty port on the D-Link USB hub :man_facepalming:.

closed 2020-03-20 by julianoes