PX4/PX4-Autopilot #14440: PX4 SITL build broken with Ubuntu 16.04

julianoes, 2020-03-20:
As raised on Slack (https://px4.slack.com/archives/C0V533X4N/p1584707186358400), the build with Ubuntu 16.04 seems to fail:

I was able to reproduce this on master inside docker.

```
[319/712] Building CXX object src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/ekf_helper.cpp.o
FAILED: /usr/bin/ccache /usr/bin/c++   -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"ecl/EKF\" -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/ecl -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-associative-math -std=gnu++11 -MMD -MT src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/ekf_helper.cpp.o -MF src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/ekf_helper.cpp.o.d -o src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/ekf_helper.cpp.o -c ../../src/lib/ecl/EKF/ekf_helper.cpp
../../src/lib/ecl/EKF/ekf_helper.cpp: In member function ‘void Ekf::runYawEKFGSF()’:
../../src/lib/ecl/EKF/ekf_helper.cpp:1802:102: error: ‘isfinite’ was not declared in this scope
  if (_gps_data_ready && _gps_sample_delayed.vacc > FLT_EPSILON && isfinite(_gps_sample_delayed.vel(0)) && isfinite(_gps_sample_delayed.vel(1))) {
                                                                                                      ^
compilation terminated due to -Wfatal-errors.
[319/712] Building CXX object src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/covariance.cpp.o
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
root@0b9ee4e9527b:~/Firmware# 
```

TSC21, 2020-03-20:
Shouldn't this be using `PX4_ISFINITE()`?

dusan19, 2020-03-20:
it compiles with that simple change

closed 2020-03-20 by LorenzMeier