PX4/PX4-Autopilot #14446: Error building px4_sitl on MacOS Catalina

breathedigital, 2020-03-20:
**What is the problem?**
Unable to successfully execute ```make px4_sitl gazebo``` on MacOS Catalina
[px4_sitl_make_error_on_mac.log](https://github.com/PX4/Firmware/files/4361399/px4_sitl_make_error_on_mac.log)

**Steps to reproduce the behavior:**
1. MacOS Catalina running Python 3.7.7 with all dev tools as per this link [1].  
2. Check if indeed ```python -v ``` is pointing to the correct binary & site-packages folder.  
3. Followed this link [2] and checked-out the latest release.
4. Ran ```make px4_sitl gazebo``` as described in [2].    

``` git checkout v1.9.2```
``` make px4_sitl gazebo```

[1] - https://dev.px4.io/v1.9.0/en/setup/dev_env_mac.html
[2] - https://dev.px4.io/v1.9.0/en/setup/building_px4.html

**Expected behavior**
PX4 SITL would have been built for Gazebo.

**Log Files and Screenshots**
Attaching the make output.  

Add screenshots to help explain your problem.
[image]

**Additional context**
I am trying to run PX4 SITL on either Gazebo or jMavSIM, but none of these targets are being built from this repository on MacOS Catalina.

julianoes, 2020-03-21:
This is only a compile warning. You should be able to compile it by removing `-Werror` from:
https://github.com/PX4/Firmware/blob/v1.9.2/cmake/px4_add_common_flags.cmake#L66

LorenzMeier, 2020-03-21:
We are focused at PX4 of always providing a latest stable release and we care about the developer experience of that and master and appreciate feedback.

Thank you for the really detailed issue report and the time you've put into accurately describing the issue. This is extremely helpful!

Unfortunately, in order to not spread the volunteer resources of our project too thin, we're not supporting to build old releases with new toolchains (v1.10 is the latest stable release), v1.9.x is now coming up to being a year old. I would strongly recommend to upgrade to 1.10, as you're missing safety-relevant features and improvements. You can build 1.10 or master with Catalina just fine. I'm closing this issue consequently.

closed 2020-03-21 by LorenzMeier

LorenzMeier, 2020-03-21:
However, feel free to send a pull request with the compile fixes for it!