PX4/PX4-Autopilot #14449: review HIL

dagar, 2020-03-20:
"when i run HITL simulations on latest master, i get sensor_baro:0 failed: TIMEOUT!, reconfiguring, messages. The attitude estimation (while landed) doesnt work, also the position and altitude.
I am usually runing HITL simulations on 1.10 release where it works fine."

Steps to reproduce
 - uploaded master to the fcu (hitl enabled, hil x airframe set)
 - compiled gazebo, changed iris model to use serial and enabled hitl
 - run gzserver

dagar, 2020-03-20:
@dusan19

LorenzMeier, 2020-03-21:
Can we get a "make abc" target for this?

dagar, 2020-03-22:
> Can we get a "make abc" target for this?

Yes. We should also look for a way to set some of these things at runtime so we don't end up with more copied gazebo models for every single possible setup.

LordHelmchen324, 2020-05-07:
I think this may be related to #14721.

After pulling the latest master and maybe doing some other updates (unfortunately I'm really not sure when it changed) I now, too, get the message `sensor_baro:0 failed: TIMEOUT!, reconfiguring priorities`.

dagar, 2020-05-07:
Thanks for the update @LordHelmchen324. I'd like to get this fixed for the upcoming v1.11 release.