PX4/PX4-Autopilot #14459: EKF altitude estimation diverges and jumps

parakhm95, 2020-03-23:
We have been noticing a bug with our drones where the EKF altitude estimate jumps by 2-3 metres and sometimes by hundreds of metres when it looks like both the GPS and Baro are reporting correctly. I am attaching three logs of such instances with description. Only thing I have noticed are EKF GPS hgt timeouts but they don't always happen at the time of this divergence and behavior. I don't know what causes the EKF GPS timeouts and I am still looking. 

In a position hold mode or position mode, the drone was not supposed to suddenly gain altitude but even without the pilot's input, it jumps. 

1. https://review.px4.io/plot_app?log=e1d4e927-2ca5-451b-bb67-275529703e03
At 4:44.500, you will see a sudden change in fused altitude and a spike in actuator control 03. 

2. https://review.px4.io/plot_app?log=7dcad278-b906-4bf1-ab47-7f2ac46e6239
On this one, you will notice this same behavior at 23:14 but this time, it completely diverges and leads to another control output spike. 

3. https://review.px4.io/plot_app?log=49045cbe-f0be-4369-a9e2-34ad874a1a8b
On this one, this behavior reappears at 2:57 and leads to a sharp rise in control output. 

On all these logs, this jump and divergence happens multiple times but doesn't always affect the drone. In some cases, it happened on the ground and so it went unnoticed.

We fly a heavy lift quadrotor drone with total take-off weight of 24.9 kgs. In this setup, it was flying empty at 18 kgs. The flight board is a Holybro Pixhawk 4 with stock GPS module shown here https://shop.holybro.com/pixhawk-4-gps-module_p1094.html. This is the one it shipped with.

CarlOlsson, 2020-03-23:
Just curious, why do you use GPS as primary height source? Do you have the same problem if you use baro instead?

If you could reproduce the error while collecting a [ekf2 replay logfile](http://dev.px4.io/v1.9.0/en/debug/system_wide_replay.html#ekf2-replay) it would be much easier to debug

parakhm95, 2020-03-24:
I checked the logs from when we used the barometer sometimes, and I have noticed jumps but not a very high divergence. Below attached logs are from when we flew using barometer : 
https://review.px4.io/plot_app?log=aef40cc4-2175-4201-b6e6-e84b91b7ba91
https://review.px4.io/plot_app?log=ee7eea86-b78a-4a18-bb64-ef235c7c0774
https://review.px4.io/plot_app?log=1c4434ad-0214-4e0a-b320-62c4c0af3ed1

I also found another log with GPS where the EKF completely diverges and goes to the stratosphere somehow. The log file is big and I am having trouble attaching it so here's a screenshot :
[image]

The reason we don't use GPS is because our propellers disturb the measurements a lot and when moving laterally, we lose a lot of altitude due to decrease in pressure. 

In response to your EKF2 replay logfile, do you say that I do EKF2 replay for the GPS logs I attached in the original comment or do I change the parameter in those logs to use baro and then do EKF2 replay?

CarlOlsson, 2020-03-24:
> The reason we don't use Baro is because our propellers disturb the measurements a lot and when moving laterally, we lose a lot of altitude due to decrease in pressure from fast moving air.

I have not tested it myself but ecl has a feature to address exactly that
https://github.com/PX4/ecl/blob/870ec380f13707cde0e6adda5bf52838aa602ebd/EKF/ekf_helper.cpp#L628-L628

 > In response to your EKF2 replay logfile, do you say that I do EKF2 replay for the GPS logs I attached in the original comment or do I change the parameter in those logs to use baro and then do EKF2 replay?

You need to set the SDLOG_PROFILE parameter to 2 and then collect a logfile where you reproduce the problem.

Note that to collect a usable replay logfile you need to use a good SD card, e.g [SanDisk Extreme U3 32GB](http://dev.px4.io/master/en/log/logging.html#sd-cards), set the [logging buffer above 40](https://github.com/PX4/Firmware/blob/1f4b2d1d53c48f8041152dd2c906502062bbad2f/ROMFS/px4fmu_common/init.d/rcS#L36-L36) and have CPU load below 70% (on a v4 processor).
Since you are using F7 you can probably set the logging buffer much higher

parakhm95, 2020-03-24:
We were trying the wind speed estimation parameters but we haven't tuned them. We don't want to operate using barometer though because we don't have wiggle room for failures with such a heavy aircraft. 

We are currently logging everything except 7: Computer Vision and Avoidance so whatever you need should be in there. We are already using a U3 V30 card but we didn't set the logging buffer higher. Can the buffer only be changed using the cpp file or is there a parameters in QGC as well? I couldn't find any.

CarlOlsson, 2020-03-24:
> We don't want to operate using barometer though because we don't have wiggle room for failures with such a heavy aircraft.

My experience with GPS altitude using standard M8N modules is that it is very unreliable so I would be careful with assuming that using GPS instead of baro will solve your problems

> We are currently logging everything except (7: Computer Vision and Avoidance) so whatever you need should be in there.

Yes but you have to many logging drops in the logfile, so it is more about removing thins you don't need

> Can the buffer only be changed using the cpp file or is there a parameters in QGC as well? I couldn't find any.

AFAIK only from the rc script

However, I did replay your logfile anyway. The logging drops makes the estimator diverge a couple of times but it is still clear that something seems broken with the mode of using GPS altitude as primary height source

baro vs GPS as primary height source
[image]

[image]

parakhm95, 2020-03-24:
> My experience with GPS altitude using standard M8N modules is that it is very unreliable so I would be careful with assuming that using GPS instead of baro will solve your problems
Do you recommend any particular GPS modules that can prove more reliable? I will start thinking about wind estimation so as to improve the state estimation.

> Yes but you have to many logging drops in the logfile, so it is more about removing things you don't need
On this subject, is there a SDLOG_PROFILE that includes everything or is it advisable to continue doing 127(01111111)? I will plan a flight and get back to you with a log file without dropouts. I will increase the log buffer also. I guess I will go ahead and build for the commit corresponding to v1.10.1 tag. 

> However, I did replay your logfile anyway. The logging drops makes the estimator diverge a couple of times but it is still clear that something seems broken with the mode of using GPS altitude as primary height source
I am currently reading [ECL EKF Overview](https://docs.px4.io/v1.9.0/en/advanced_config/tuning_the_ecl_ekf.html). Any other leads to start digging into?

CarlOlsson, 2020-03-24:
> Do you recommend any particular GPS modules that can prove more reliable?

No, but it is also very hard without knowing your application etc. There are very good GPS modules, but they might set you back >5K CHF

> On this subject, is there a SDLOG_PROFILE that includes everything or is it advisable to continue doing 127(01111111)?

Why did you exclude the default set? No, you might need to create your own one if you need something specific

> I am currently reading ECL EKF Overview. Any other leads to start digging into?

Not really, very hard to say where the bug is without digging into the code

parakhm95, 2020-03-24:
> Why did you exclude the default set? No, you might need to create your own one if you need something specific. 

We wanted to have all the data possible(we shifted to pixhawk recently) in case we crash our drone so that we can study it. 

I used a small F450 frame just now with HGT mode set to GPS and SDLog configured properly and it happened again. This time, it lost altitude instead of gaining altitude. And when looking through plotjuggler it is clear that EKF estimate jumped somehow.

CarlOlsson, 2020-03-24:
> We wanted to have all the data possible(we shifted to pixhawk recently) in case we crash our drone so that we can study it.

Yes, so why did you **exclude** the default set, now you are missing some topics.

> I used a small F450 frame just now with HGT mode set to GPS and SDLog configured properly and it happened again. This time, it lost altitude instead of gaining altitude. And when looking through plotjuggler it is clear that EKF estimate jumped somehow.

The issue has to do with the mess around GPS checks, different but similar to the one fixed [here](https://github.com/PX4/ecl/pull/761). 

The bug was introduced here
https://github.com/PX4/ecl/pull/614

Without commit/with commit
[image]

As soon as your GPS quality checks are failing it takes 10 seconds for the EKF to declare them good again. But the timeout for when to switch to baro altitude is 5 seconds so that is why it switches back and forth.

The solution to this will have to be a bigger discussion regarding what the GPS checks actually represent (only preflight or general measurement quality). 
In the meantime if you want to keep flying with GPS as primary altitude source you will have to tune the GPS quality check thresholds defined around `EKF2_GPS_CHECK`.

CarlOlsson, 2020-03-24:
opened an issue on ecl repo https://github.com/PX4/ecl/issues/786, thanks for reporting!

parakhm95, 2020-03-24:
Thank you so much! I had my doubts about this from the beginning. We were struggling to get the checks to pass a few days ago and I set the EKF2_REQ_SACC higher because it looked it was failing that exact check. I have further questions about this if you could answer please : 

1. If I have to verify these parameters for GPS checks, where should I check? I have been checking the vehicle_gps_position and ekf2_gps_drift.
2. Should an EKF diverge from both of its sensor values when it switches? Is this unusual behavior for an EKF? 
3. How do I see these EKF gps hgt timeout warnings during the flight? Is there a topic I can listen to? 

> Yes, so why did you exclude the default set, now you are missing some topics.

I selected everything apart from bit 7 and I didn't expect to be **excluding** instead of including when I selected them. I am kinda confused about how I managed to miss things by checking more boxes. Can you please clear this up for me? 

> The solution to this will have to be a bigger discussion regarding what the GPS checks actually represent (only preflight or general measurement quality).
In the meantime if you want to keep flying with GPS as primary altitude source you will have to tune the GPS quality check thresholds defined around EKF2_GPS_CHECK.

Is this debate happening [here](https://github.com/PX4/ecl/pull/761) ?

parakhm95, 2020-03-24:
Never mind about the last question I got my answer.

CarlOlsson, 2020-03-25:
> I selected everything apart from bit 7 and I didn't expect to be excluding instead of including when I selected them. I am kinda confused about how I managed to miss things by checking more boxes. Can you please clear this up for me?

I don't know, I must have been confusing things yesterday xD You are right, you have also selected the default set of topics..

> If I have to verify these parameters for GPS checks, where should I check? I have been checking the vehicle_gps_position and ekf2_gps_drift.

The GPS checks are done [here](https://github.com/PX4/ecl/blob/master/EKF/gps_checks.cpp#L116). To know which check fail you can look in `estimator_status.gps_check_fail_flags`

> Should an EKF diverge from both of its sensor values when it switches? Is this unusual behavior for an EKF?

The altitude measurements from GPS and baro will always have an offset between them. However, the EKF should handle this. But from the plots above it looks like something isn't really right there, also looking at the vertical position innovations right after the switch, at least for fast switching back and forth. I don't have time to look into this but I would start the investigation [here](https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L1032-L1051) and e.g. add `_hgt_sensor_offset` to logging

> How do I see these EKF gps hgt timeout warnings during the flight? Is there a topic I can listen to?

You can see which sensor the EKF is using as height source in `estimator_status.control_mode_flags`

> Also, I am unsure if EKF2_REQ_SACC refers to the correct parameter because its default is set as 0.5 but GPS_PERFORMANCE_REQUIREMENT mentions the wrong default value. And yet, changing it helps the fix somehow so I am puzzled.

That seems to be wrong in the docs. Can you please raise an issue on the [user guide repo](https://github.com/PX4/px4_user_guide/issues)?

closed 2022-08-15 by bresch