PX4/PX4-Autopilot #14464: Fail to use external Magnetometer

DavidBSaba, 2020-03-24:
Hello

I am trying to use the external magnetometer as the primary source for heading calculation. I am doing this in order to reduce the magnetic field interference from power cables.

I am currently using Pixhawk 2 cube and the PX4 software (version 1.8.0 stable). I connected a Here+ module to the GPS1 port.

I tried all possible combinations of the values of the parameters 
CAL_MAGx_EN, x=0,...,3
CAL_MAGx_ROT
CAL_MAG_PRIME

The tests mainly consisted of disabling the Mag2 (which supposed to be the internal Mag) and enabling the Mag0 or Mag1 (witch at least one of them is supposed to be external), from the GS and directly from the code and then changing the CAL_MAG_PRIME in order to coincide with the ID of Mag0 or Mag1. 
(https://discuss.px4.io/t/custom-yaw-rotation-of-extrernal-magnetometer/9562)

The yaw was always following the internal magnetometer (the yaw indicated in the GS follows the orientation of the controller), and I never succeeded to make him follow the orientation of the GPS module (containing the external Mag). Plus, after calibration, all the parameters CAL_MAGx_ROT (x=0,...,3) pass automatically to 'Internal', regardless of the value I set in the firmware or in the GS before calibration.

Can anyone please help me solving this issue?

Thank you

julianoes, 2020-03-26:
I don't think anyone can go back to 1.8.0 to help you with this issue. I suggest to try newer releases and check if it has been fixed in the meantime. Sorry about that.

DavidBSaba, 2020-04-02:
Thank you @julianoes. I passed to version 1.10.1.
The external magnetometer is working fine

julianoes, 2020-04-02:
Nice thanks for the feedback @DavidBSaba

closed 2020-04-02 by julianoes