PX4/PX4-Autopilot #14465: Kill switch for companion computer

cimbar, 2020-03-24:
Hi,
In 1.8.2 release and before, disarming with mavlink message was possible. I was using mavros and sending mavros/State message as armed=false and it was working as same as manual kill switch. While flying in offboard mode with companion computer and not using RC controller, this was an option for emergency stop. When i passed to 1.10 version, i realized that this option is not possible anymore and returns this error: 
FCU: Disarming denied! Not landed
If i am missing some parameter to change, i am sorry but if there isn't any possibility for emergency stop via mavlink message, i think it should be implemented.
Thanks

TSC21, 2020-03-24:
You need to send the arm command with the param 2 set to `21196`. https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM. You can use the [`mavcmd`](https://github.com/mavlink/mavros/blob/master/mavros/scripts/mavcmd) for that same purpose.

TSC21, 2020-03-24:
You can also extend [`mavsafety`](https://github.com/mavlink/mavros/blob/master/mavros/scripts/mavsafety) with that option, and its actually the best way of doing it. If you do it so, please issue a PR to MAVROS, so everyone can benefit from it.

cimbar, 2020-03-25:
Following your suggestion, i used mavros_msgs/CommandLong service on /mavros/cmd/command topic inside of the code instead of using mavcmd. Also, users can call service from bash with
```
rosservice call /mavros/cmd/command "{broadcast: false, command: 400, confirmation: 0, param1: 0.0, param2: 21196.0, param3: 0.0,
  param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0}"
```
Thanks for help

julianoes, 2020-03-26:
Alternatively there is also:
https://mavlink.io/en/messages/common.html#MAV_CMD_DO_FLIGHTTERMINATION
But for this the flight termination circuit breaker might need to be disabled.

TSC21, 2020-03-26:
> Following your suggestion, i used mavros_msgs/CommandLong service on /mavros/cmd/command topic inside of the code instead of using mavcmd. Also, users can call service from bash with
>
> ```
> rosservice [...]

@cimbar are you interested in extending the `mavsafety` with the feature as well? Like a `kill` command?

cimbar, 2020-03-26:
#https://github.com/mavlink/mavros/pull/1405

closed 2020-03-31 by cimbar