PX4/PX4-Autopilot #14466: Return To Home failure

NicoNelli, 2020-03-24:
Hi everyone,

I wanted to test the RTL using my VTOL Quad Tailsitter and I was on branch master. When I called the RTL, the vehicle was in FW. So the vehicle started going back and then it did the transition back to MC just before landing (till now everything was fine). The vehicle started going down (2 m) and it started drifting horizontally. I called again the RTL and it reached again the return point, it went down a bit and then it drifted again. The third times I triggered the RTL everything was fine and the vehicle landed. I repeated this several time and always the same behaviour. 
Could you please give me some advices about this strange behaviour ?
Thanks to everyone, I also shared here the logs.  

https://logs.px4.io/plot_app?log=5056e92a-e9a8-47bd-9581-34669d4f7158

julianoes, 2020-03-26:
FYI @RomanBapst

RomanBapst, 2020-03-26:
@NicoNelli I think you have a problem with your remote control. The channel mapped to RTL glitches once in a while and would explain why the vehicle goes out of RTL.
[image]

closed 2020-03-26 by RomanBapst

NicoNelli, 2020-03-26:
Thanks @RomanBapst for the quick reply. I did that in purpose. 
The very first time I called the RTL i saw the drone drifting. 
Based on that I said: let's try to call again and again the RTL and see what happens.
Accordingly I did again the test and after calling three times the RTL the vehicle was able to land successfully.

RomanBapst, 2020-03-30:
@NicoNelli So is there still an issue or can we keep this closed?

NicoNelli, 2020-03-31:
@RomanBapst there is still the issue. 
Maybe I did not explain it in a very good way but the spikes that can be seen in the RTL channel actually were triggered by me. During a flight test, I called the RTL and the drone started going back and when it reached the landing point, it descended for ~ two meters and it drifted. I retriggered again the RTL and same behaviour. The third time I triggered the RTL the drone landed successfully.
[image]
[image]

Looking at the local positions x and z for instance, we can see that after I triggered the RTL the drone went down and then it drifted. The third time I triggered the RTL the drone landed.

NicoNelli, 2020-04-11:
@RomanBapst do you have suggestions in that matter ? Thanks a lot

dayjaby, 2020-04-11:
I observed the same fault (https://discuss.px4.io/t/drone-flies-away-during-rtl/15373). Firmware version: 1.9.2 with slight modifications (but nothing RTL related) on a hexacopter (no vtol!)

Which firmware version did you use?

[image]

dagar, 2020-04-11:
If there's still a problem here could you please validate against current PX4 and open a new issue with logs?

dayjaby, 2020-04-11:
@dagar In our case it stopped occuring in a long time and we had now several months of good test flights on the same firmware version, the same location. On some days however (~october 2019), it happened frequently.

In my log, it didn't drift out of RTL, but the estimation drifted away (and also the drone in reality), even though the setpoint remained constant.

NicoNelli, 2020-04-14:
@dagar ok I will do it. Thanks a lot

dayjaby, 2020-04-16:
https://github.com/PX4/Firmware/issues/13956 has the same kind of drift during RTL.

dagar, 2020-04-16:
> #13956 has the same kind of drift during RTL.

Thanks @dayjaby. I've added https://github.com/PX4/Firmware/issues/13956 as a v1.11 release blocker. Any additional data or testing you can add to the issue usually doesn't hurt.