PX4/PX4-Autopilot #14479: Battery emergency RTL re-initiate landing command, even if drone already lands

BazookaJoe1900, 2020-03-25:
**Describe the bug**
I found on a flight that I have came to nearly empty battery that there is something that look to me like bug in the emergency and critical battery logic.
I started to land in nearly empty battery, but then the emergency battery logic intiated, and instead of continue landing, it started to RTL (climbing), I took control to RC and landed.
it seems that COM_LOW_BAT_ACT doesn't respect the current flight mode, and if the drone has already started to land, it can re-initiate action, that might be climbing.

**To Reproduce**
 * in order to reproduce easily in simulator I have changed the next parameters:
BAT_LOW_THR = 30%
BAT_EMERGEN_THR  = 20%
BAT_CRIT_THR = 10%
COM_LOW_BAT_ACT = 3 (Return mode at critically low level, Land mode at current position if reaching dangerously low levels)
RTL_RETURN_ALT = 100m
SIM_BAT_MIN_PCT = 0%

 * take-off
 * fly to another location
 * climb to 20m
 * wait that the battery will get to ~30% and start land
 * the RTL will be initiated, starting with climbing (and here he bug)

**Expected behavior**
in case of already started landing, don't initiate RTL. and don't re-initiate landing

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=c50117f1-3f81-4141-af5f-5e905ea239e5
[image]

**Drone (please complete the following information):**
- simulation (master 25/3/2020)

BazookaJoe1900, 2020-04-19:
@julianoes Is there someone I can refer this issue?
I am not that familiar with this sections of code, and its critical code, so I don't want want to mess things there.

julianoes, 2020-04-20:
@RomanBapst might have a better idea.

BazookaJoe1900, 2020-07-19:
still relevant

closed 2021-01-10 by LorenzMeier