PX4/PX4-Autopilot #14500: RDDRONE-UAVCAN146 Link order issue

PetervdPerk-NXP, 2020-03-27:
When enabling NuttX buttons and leds I get the following linker error

```
NuttX/nuttx/drivers/libdrivers.a(button_lower.o): In function `btn_disable':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/input/button_lower.c:188: undefined reference to `board_button_irq'
NuttX/nuttx/drivers/libdrivers.a(button_lower.o): In function `btn_enable':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/input/button_lower.c:162: undefined reference to `board_button_irq'
NuttX/nuttx/drivers/libdrivers.a(button_lower.o): In function `btn_buttons':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/input/button_lower.c:115: undefined reference to `board_buttons'
NuttX/nuttx/drivers/libdrivers.a(button_lower.o): In function `btn_lower_initialize':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/input/button_lower.c:233: undefined reference to `board_button_initialize'
NuttX/nuttx/drivers/libdrivers.a(userled_lower.o): In function `userled_ledset':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/leds/userled_lower.c:120: undefined reference to `board_userled_all'
NuttX/nuttx/drivers/libdrivers.a(userled_lower.o): In function `userled_led':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/leds/userled_lower.c:106: undefined reference to `board_userled'
NuttX/nuttx/drivers/libdrivers.a(userled_lower.o): In function `userled_lower_initialize':
/home/hovergames/nuttx/PX4-Firmware/build/nxp_rddrone-uavcan146_default/NuttX/nuttx/drivers/leds/userled_lower.c:138: undefined reference to `board_userled_initialize'
collect2: error: ld returned 1 exit status
```

I can solve this manually by re-ordering LINK_LIBRARIES in the build/build.ninja file, however this file is dynamically generated. 

This problem is caused by libdrivers.a searching for symbols located in libdrivers_board.a. by re-arranging these libraries in the correct order this problem is solved. Unfortunately I'm not familiar enough with cmake and the PX4 build system to find the culprit of the issue in CMake configuration files.

dagar, 2020-03-27:
On the PX4 side of the build the dependencies are (mostly) explicitly handled these days. Where the build touches NuttX it's a bit of a mess. To resolve all the circular dependencies the NuttX libraries are linked with start-group and end-group.

As an initial test try adding `drivers_board` to within the NuttX start-group/end-group here:
https://github.com/PX4/Firmware/blob/e5dd2e7a1929f1cdadc76f4f3de68a73296a72ad/platforms/nuttx/CMakeLists.txt#L118-L120

PetervdPerk-NXP, 2020-03-27:
By adding `drivers_board` within the NuttX start-group/end-group it will compile/link successfully
```
	-Wl,--start-group
		drivers_board
		${nuttx_libs}
	-Wl,--end-group
```

dagar, 2020-03-27:
Great, if you want to make the change upstream we could also try adding it as an explicit `nuttx_drivers` linking dependency.

https://github.com/PX4/Firmware/blob/e5dd2e7a1929f1cdadc76f4f3de68a73296a72ad/platforms/nuttx/CMakeLists.txt#L93-L97

closed 2020-03-27 by dagar