PX4/PX4-Autopilot #14505: lsm303d & l3gd20 can not calibrate level horizon

rolandash, 2020-03-27:
**Describe the bug**
lsm303d & l3gd20 can not be calibrate leveled in single calibration process.
This issue should infect all FCUs have these 2 sensors onboard (>= v1.10).
This issue can only be visible/detected when FCU has only these 2 sensors on-board (or uses these 2 as primary motion sensors?)
This issue does not exist in earlier version of version (<=v1.9).
This issue is not observed with other type of IMU sensors so far.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a FCU that have lsm303d & l3gd20 on board. If you are using MindPX, remove the isolated redundant IMU module. If you are using Pixhawk, remove other sensors on-board. 
2. connect to QGC. 
3. Clear all parameters to start over.
4. Select a generic airframe, like Generic Quad X.
5. Perform a complete sensor calibration process, util 'level horizon' finished and all indicators becomes green.
6. Place FCU in level, go to fly view and check the flight gadget. The attitude gadget is not leveled. 

**Expected behavior**
 The attitude gadget should be leveled.

**Additional context**
The issue can be corrected by a second 'level horizon' action. 
1. Following above step 6, do a 'level horizon' action again. 
2. reboot the FCU and connect to QGC.
3. The attitude gadget is leveled this time.

dagar, 2020-03-27:
Would you be able to get a log from boot (SDLOG_MODE 2) with an attempted calibration?

Capturing this output from the console might also help.

`sensors status`

`listener sensor_accel`

`listener sensor_gyro`

`work_queue status`

rolandash, 2020-03-28:
@dagar log below as you requested.

**Update**
I have tested that disabling other sensor's driver is equivalent to physically remove them from board. So if you got a Pixhawk, then you can reproduce it simply by disable other motion sensor's driver and only left lsm303d & l3gd20 there.

````
FMUv2 ver 0x8 : Rev 0 V2
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 738c3939a796bd70216763246a6ec432ea7d4f90
FW version: 1.11.0 0 (17498112)
FW git-branch: pr-new_invensense_drivers
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: b902d4ca3d79ffbd5148670cb13bfeef7d917555
Build datetime: Mar 27 2020 18:05:56
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000100000000373936353235510a004a0026
MCU: STM32F42x, rev. 3
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 16
Board defaults: /etc/init.d/rc.board_defaults
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
rgbled #0 on I2C bus 2
ms5611 #0 on SPI bus 1 (devid=0x3e)
l3gd20 #0 on SPI bus 1 (devid=0x22)
lsm303d #0 on SPI bus 1 (devid=0x11)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: /fs/microsd/log

NuttShell (NSH)
nsh> 
nsh> INFO  [commander] [cal] calibration started: 2 mag
INFO  [commander] [cal] pending: back front left right up down
INFO  [commander] [cal] hold vehicle still on a pending side
INFO  [commander] [cal] detected rest position, hold still...
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] Rotate vehicle around the detected orientation
INFO  [commander] [cal] Continue rotation for down 7 s
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] Rotate vehicle around the detected orientation
INFO  [commander] [cal] Continue rotation for down 7 s
INFO  [commander] [cal] down side calibration: progress <4>
INFO  [commander] [cal] down side calibration: progress <8>
INFO  [commander] [cal] down side calibration: progress <12>
INFO  [commander] [cal] down side done, rotate to a different side
[... 405 lines cut ...]
|__ 2) wq:hp_default   
|   |__ 1) tone_alarm                   10.2 Hz      98469.8 us
|   |__ 2) rc_update                     0.0 Hz          0.0 us
|   |__ 3) adc                         100.0 Hz       9999.8 us (10000 us)
|   |__ 4) battery_status              100.0 Hz       9999.8 us (10000 us)
|   \__ 5) fmu                           0.0 Hz  195050592.0 us
|__ 3) wq:I2C2         
|   \__ 1) rgbled                       38.1 Hz      26225.4 us
|__ 4) wq:SPI1         
|   |__ 1) ms5611                       99.5 Hz      10054.1 us
|   |__ 2) l3gd20                     1398.6 Hz        715.0 us (715 us)
|   \__ 3) lsm303d                     952.4 Hz       1050.0 us (1050 us)
|__ 5) wq:att_pos_ctrl 
|   |__ 1) sensors                     218.9 Hz       4567.5 us
|   |__ 2) vehicle_acceleration        777.8 Hz       1285.6 us
|   |__ 3) vehicle_air_data             74.6 Hz      13406.5 us
|   |__ 4) vehicle_imu                 428.7 Hz       2332.5 us
|   |__ 5) ekf2                        218.9 Hz       4567.5 us
|   |__ 6) mc_att_control              185.0 Hz       5406.7 us
|   |__ 7) mc_pos_control               33.2 Hz      30124.2 us
|   \__ 8) land_detector                50.0 Hz      19999.3 us (20000 us)
\__ 6) wq:rate_ctrl    
    |__ 1) vehicle_angular_velocity    742.7 Hz       1346.5 us
    \__ 2) mc_rate_control             323.6 Hz       3089.8 us
````

rolandash, 2020-03-30:
@dagar I need to add an update from my testings, might be helpful:
Under v1.11.0 mpu6500 seems also have the same behavior as lsm303d & l3gd20. mpu6k works good.
Under v1.10.0 I use my previous PR of 9250 driver for mpu6500 and it works just fine.
So seems like a reasonable guess that there's something not right with the new driver.

dagar, 2020-04-08:
@rolandash I believe the problem here is the estimator not initializing with the data from the lsm303d + l3gd20 combo. Does it work if you do a clean accel + gyro calibration, reboot, then try level horizon?

rolandash, 2020-04-10:
@dagar My testings confirms with your judgement. 
If I do level horizon after sensor cal + reboot, then it works fine. 

Expecting the fix from you soon.

dagar, 2020-05-07:
@rolandash could you please give current master another try? It's about to be tagged as v1.11.0-beta2. If it's still a problem a log from boot (SDLOG_PROFILE) would help.

rolandash, 2020-05-12:
@dagar Just saw your message. Will test it and feedback to you in one day or two.

rolandash, 2020-05-15:
@dagar
unfortunately the problem seems still around In my testings. I will upload logs later soon.

rolandash, 2020-05-16:
@dagar

Please check the SDLOG attached.

[log001.zip](https://github.com/PX4/Firmware/files/4638823/log001.zip)

rolandash, 2020-08-18:
Hi @dagar 

Is this issue considered been addressed in latest 1.11 ?

dagar, 2020-08-18:
> Hi @dagar
> 
> Is this issue considered been addressed in latest 1.11 ?

Well yes and no.

It appears to be an issue with ekf2 where if the IMU has a sufficiently large offset it won't initialize. If you calibrate first, the estimator will initialize (and start publishing vehicle_attitude) and the level horizon calibration will then work. 

We could open a PX4/ecl bug and investigate it there, but I'm wondering if we simply need a better error message to suggest calibrating before trying again.

ryanjAA, 2020-08-30:
I loaded up 1.11 RC3 zeroed all accel, gyro and mag calibrations and calibrated all in one go including leveling the horizon and it worked. Sensors onboard were LSM303D for compass (1114122), MPU600 for gyro (2162698). Gyro 2 as L3GD20 (2228234). Offsets would have been zero since I did it all from zero but it does work. Soft reboot and leveled again and leveled fine. No external sensors. All internal.

ryanjAA, 2020-12-29:
So this is interestingly back. On release 1.11.2 if i follow the above and calibrate gyro and accel first, i can calibrate the horizon but once i reboot i cant calibrate it. Looking at the offsets of the accels, one of them has offsets of around 3 (MRO Pixhawk and i tried 3 different ones). 

Presumably all hardware will have an offset different between two different sensors so this not being able to cope with that is probably the change needed.

At the same time it isn't really feasible to have to calibrate gyro and accel every time you want to level the horizon (mag doesn't seem to matter here).

The shortcut I've been using is if you hit calibrate on the horizon and let it sit for a few seconds, it will get to the end, just not complete. Then exiting out the calibration pane (not hitting cancel) and checking the Sens board offset has the new offsets in there. That seems to get the level horizon done as a hack.

dagar, 2020-12-29:
It should be better in master now (what will be v1.12), I relaxed the acceleration threshold slightly for ecl/EKF initialization not too long ago.

ryanjAA, 2020-12-29:
OK Cool - Will the hack im currently doing be sufficient though for leveling in the mean time?

dagar, 2020-12-29:
> OK Cool - Will the hack im currently doing be sufficient though for leveling in the mean time?

I would say yes only if you're not having to redo it more than once per vehicle.

What's happening is the lsm303d + l3gd20 combo is being given the same initial priority as the mpu6000, but since it has a much faster reset it ends up publishing data first and becomes the selected sensor. I would either lower the priority of the lsm303d + l3gd20 (CAL_ACCx_PRIO/CAL_GYROx_PRIO) or in startup add a delay between starting the mpu6000 and lsm303d + l3gd20 so that you get the mpu6000 first (which has much smaller offsets).

All of this becomes irrelevant with Multi-EKF.

ryanjAA, 2020-12-29:
I'll try lowering the priority right now.

It happens to every vehicle on 1.11.2 after every reboot.

Excited about multi-ekf, just want to get this useable in the mean time.

ryanjAA, 2020-12-29:
Ah - 1.11.2 doesn't have sensor priority yet.

Ok so how to delay them and by how much?

LorenzMeier, 2020-12-30:
@dagar This sounds though like the condition for sensor usage in EKF2 should be reset once the system goes out of calibration mode.

dagar, 2020-12-30:
> @dagar This sounds though like the condition for sensor usage in EKF2 should be reset once the system goes out of calibration mode.

It already works that way, it's just really unintuitive if you fall into that hole because there's nothing that directly connects the underlying problem and what to do about it.
 -  ecl/EKF refuses to initialize (silently) because the accel norm exceeds the threshold => level horizon fails

To sidestep the issue I bumped the threshold a few weeks ago. https://github.com/PX4/PX4-ECL/blob/18f334f4dbca61e3bcd1743c22cf820b5a8ff7ef/EKF/ekf.cpp#L239-L240

The vast majority of sensors we support are nowhere need bad enough to have this problem.

ryanjAA, 2020-12-30:
If ecl/ekf is refusing to initialize is this a problem for flight or only for sensor calibration?

Meaning if the horizon doesn’t need to be leveled again as it was done previously are there concerns ekf still won’t initialize?

dagar, 2020-12-30:
> If ecl/ekf is refusing to initialize is this a problem for flight or only for sensor calibration?

It's only a problem at initialization/startup, and only for particularly bad sensors old sensors.

ryanjAA, 2020-12-30:
Thanks. Was worried for a moment to fly with them.

ryanjAA, 2020-12-31:
As @dagar pointed out here https://px4.slack.com/archives/C0V533X4N/p1609285210239700?thread_ts=1609283837.239400&cid=C0V533X4N adding a sleep 1 to the line fixes this but just remember that when you startup you need to wait ~3 seconds before leveling the horizon or you will get "attitude estimator not running - check system boot”. Attached is the binary compiled (and compressed to GitHub accepts the file type) so you can just flash it. No other changes made and built from 1.11.2 release
[px4_fmu-v3_default.px4.zip](https://github.com/PX4/PX4-Autopilot/files/5758198/px4_fmu-v3_default.px4.zip)

ryanjAA, 2021-03-08:
Could this somehow be related to how I'm connected?

When I try and level the horizon via USB it works fine but when wireless it does not.

@dagar

ryanjAA, 2021-03-08:
So if I set MAV_0_rate to 1200 it doesn't even finish (when connected wirelessly). When I set it to 0 (aka half of max), the vehicles beeps that it is complete but QGC hangs at the very end of the cal.