PX4/PX4-Autopilot #14509: u-Blox Neo-M9N GPS support

eli-IF, 2020-03-27:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
I ordered an mRo GPS that was listed as supported but it came with an u-Blox Neo-M9N chip and it is not plug and play. I used "gps status" in the mavlink console and it shows that gps status is not ok.

**Describe your preferred solution**
I would prefer to have a plug and play solution for this specific GPS. I assume this requires changing the driver for gps but I am not familiar enough to do this myself.

dagar, 2020-03-27:
@bkueng any ideas here?

dagar, 2020-03-27:
@pkocmoud FYI

pkocmoud, 2020-03-28:
Form Jordi: 
So M9N doesn't support the UBX->SOL message anymore. 
The new message that has to be enabled is the UBX->PVT

LorenzMeier, 2020-03-28:
@dagar Could you please assign Beat to issues for anything GPS related? We need to get better at assigning and follow up. Thanks!

ryanjAA, 2020-04-15:
Took a little trickery for some tests but pretty incredible unit after some indoor and outdoor testing. 29 satellites outside with not a great shot to the sky and 24 inside nowhere near any windows. 

[image]

dagar, 2020-05-01:
This is included in v1.11.

ryanjAA, 2020-05-01:
I don’t think this should be closed yet. It doesn’t work yet. 

If someone else wants to test an M9N and confirm that it works that would be great but as of yesterday we tested it and in the current state is still just gives a timeout without functionality. See https://github.com/PX4/GpsDrivers/pull/57#issuecomment-621639910 more more details.

dagar, 2020-05-01:
> I don’t think this should be closed yet. It doesn’t work yet.

Thanks for the update. Let's try to keep conversation going outside of merged PRs so it isn't lost.

bkueng, 2020-05-04:
@ryanjAA continuing here from https://github.com/PX4/GpsDrivers/pull/57#issuecomment-623031721

> Whats the difference between this and master? If i flash master as of right now from QGC, it works. If i pull cdee27f in as i did from ~2 days ago, it does not. Normally is wouldn't matter but if this is trying to be put into 11 release this needs to work with that and not master.

Can you be more specific with the repos and commits you use? You mean Firmware `master` (which commit)? This will all go into the release, so if it works there we're all good. There was another issue that could cause issues with GPS, which you might have hit (https://github.com/PX4/Firmware/pull/14771).

ryanjAA, 2020-05-04:
Well this is a little strange.

Master (at 8bc72fe771, https://github.com/PX4/Firmware/pull/14785) with GPS Debug settings changed that did not work as shown before (times out) is now working (see attached) but i let the unit sit (so the onboard battery maybe drained?). There isn't much else I can think of since the time out is the same (when switching bauds), it just now seems to at some point find in (near instantly) where before it was sitting for a minute or two and still not recognizing the unit. If you flash 8bc72fe771, it works instantly and then I checked by manually by turning off the driver and then back on to see the Debug timeout messages.

Master (at cd59f95, https://github.com/PX4/Firmware/pull/14788) that was pulled in from QGC does work

Master (at 70329ce, https://github.com/PX4/Firmware/pull/14726) works with 4cc28b8, https://github.com/PX4/GpsDrivers/pull/55 (GPS Driver pulled in and hz manually changed to 5hz to test max sat usage [meas 100 to 200]).

Beta 746b312, https://github.com/PX4/Firmware/pull/14609 (from qgc) does not work.

Master as of writing (466b5db, https://github.com/PX4/Firmware/pull/14784) pulled from QGC works (and frequency is at 5hz).

If you're happy with that, I'd say it works and close. If you want me to test against https://github.com/PX4/Firmware/pull/14771, let me know.

[image]

bkueng, 2020-05-05:
Ok, thanks for the testing. I say we leave it at that since master works, and if you notice issues (such as reliability in module detection), open a new issue.

closed 2020-05-05 by bkueng