PX4/PX4-Autopilot #14513: pause PreFlightCheck during sensor calibration

dagar, 2020-03-28:
```
INFO  [commander] [cal] hold vehicle still on a pending side
ERROR [PreFlightCheck] state 15: |0.00315654| > 0.00173000 + 0.00067082
ERROR [PreFlightCheck] state 15: |0.00316423| > 0.00173000 + 0.00067082
INFO  [commander] [cal] detected rest position, hold still...
ERROR [PreFlightCheck] state 13: |-0.00255149| > 0.00173000 + 0.00067082
ERROR [PreFlightCheck] state 13: |-0.00313897| > 0.00173000 + 0.00067082
INFO  [commander] [cal] down side already completed
INFO  [commander] [cal] pending: back front left right
INFO  [commander] [cal] hold vehicle still on a pending side
ERROR [PreFlightCheck] state 13: |-0.00317967| > 0.00173000 + 0.00067082
ERROR [PreFlightCheck] state 13: |-0.00316114| > 0.00173000 + 0.00067082
INFO  [commander] [cal] detected rest position, hold still...
ERROR [PreFlightCheck] state 13: |-0.00313994| > 0.00173000 + 0.00067082
INFO  [commander] [cal] detected motion, hold still...
ERROR [PreFlightCheck] state 13: |-0.00313484| > 0.00173000 + 0.00000000
ERROR [PreFlightCheck] state 13: |-0.00313091| > 0.00173000 + 0.00000000
ERROR [PreFlightCheck] state 13: |-0.00311079| > 0.00173000 + 0.00067084
ERROR [PreFlightCheck] state 13: |-0.00313807| > 0.00173000 + 0.00067082
ERROR [PreFlightCheck] state 13: |-0.00313651| > 0.00173000 + 0.00067082
INFO  [commander] [cal] detected rest position, hold still...
ERROR [PreFlightCheck] state 13: |-0.00313159| > 0.00173000 + 0.00067082
ERROR [PreFlightCheck] state 13: |-0.00310030| > 0.00173000 + 0.00067082
```

mrpollo, 2020-04-06:
is this something we can ask contributors to help with? @dagar 

can you provide guidance on how to fix?

dagar, 2020-05-06:
Fixed in #14825.

closed 2020-05-06 by dagar