PX4/PX4-Autopilot #14526: I2CSPIDriverBase allowing multiple instances to start

dagar, 2020-03-30:
It appears to be possible to start multiple instances of certain drivers (I2CSPIDriverBase). 

For example on a pixhawk 4 that only has a single ist8310 (no gps/compass attached).

``` Console
nsh> ist8310 status
INFO  [SPI_I2C] Running on I2C Bus 3
ist8310_read: 539 events, 457497us elapsed, 848.79us avg, min 830us max 1719us 55.323us rms
ist8310_com_err: 0 events
poll interval:  10000 interval
 sensor_mag_s
        timestamp: 7463012  (0.005813 seconds ago)
        error_count: 0
        device_id: 396825 (Type: 0x06, I2C:3 (0x0E)) 
        x: -0.2340
        y: 0.2010
        z: 0.4080
        temperature: 0.0000
        scaling: 0.0030
        x_raw: -78
        y_raw: 67
        z_raw: 136
        is_external: False

nsh> ist8310 -I start
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
ist8310 #1 on I2C bus 3

nsh> ist8310 -I startERROR [sensors] Mag #0 fail:  STALE!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Mag #1 priority: 100

IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
ist8310 #2 on I2C bus 3

nsh> work_queue status

Work Queue: 7 threads                      RATE        INTERVAL
|__ 1) wq:hp_default
|   |__ 1) safety_button                30.3 Hz      32976.3 us (33000 us)
|   |__ 2) tone_alarm                   10.4 Hz      95967.5 us
|   |__ 3) rc_update                     0.0 Hz          0.0 us
|   |__ 4) adc                         100.0 Hz       9996.9 us (10000 us)
|   |__ 5) battery_status              100.0 Hz       9995.8 us (10000 us)
|   \__ 6) fmu                           0.1 Hz   11174078.0 us
|__ 2) wq:lp_default
|   |__ 1) send_event                   29.9 Hz      33432.8 us (33333 us)
|   |__ 2) load_mon                      1.0 Hz     959158.4 us (1000000 us)
|   \__ 3) mc_hover_thrust_estimator      0.0 Hz          0.0 us
|__ 3) wq:SPI1
|   |__ 1) mpu6000                    1249.7 Hz        800.2 us (800 us)
|   |__ 2) bmi055                     1249.9 Hz        800.1 us (800 us)
|   \__ 3) bmi055                     1250.0 Hz        800.0 us (800 us)
|__ 4) wq:I2C3
|   |__ 1) ist8310                      89.3 Hz      11202.8 us
|   |__ 2) ist8310                      89.3 Hz      11201.2 us
|   \__ 3) ist8310                      89.3 Hz      11202.5 us
|__ 5) wq:SPI4
|   \__ 1) ms5611                       99.4 Hz      10059.5 us
|__ 6) wq:att_pos_ctrl
|   |__ 1) sensors                     249.4 Hz       4010.1 us
|   |__ 2) vehicle_acceleration        995.9 Hz       1004.1 us
|   |__ 3) vehicle_air_data             74.5 Hz      13424.6 us
|   |__ 4) vehicle_imu                 471.8 Hz       2119.7 us
|   |__ 5) vehicle_imu                 249.4 Hz       4009.2 us
|   |__ 6) ekf2                        249.4 Hz       4009.4 us
|   |__ 7) mc_att_control              202.6 Hz       4935.8 us
|   |__ 8) mc_pos_control               33.8 Hz      29601.3 us
|   \__ 9) land_detector                50.0 Hz      19982.7 us (20000 us)
\__ 7) wq:rate_ctrl
    |__ 1) vehicle_angular_velocity    996.0 Hz       1004.0 us
    \__ 2) mc_rate_control             996.2 Hz       1003.8 us

```

SalimTerryLi, 2020-03-30:
I've run into that as well. The solution could be [here](https://github.com/PX4/Firmware/pull/14437/commits/d7f11ad43cf9a5d3332ee116a8ac55c1152d5e18)
Compared to [here](https://github.com/PX4/Firmware/blob/a38d4192ab6b9a3aae2d66fc2972ca7fa8c353a2/src/drivers/magnetometer/ist8310/ist8310.cpp#L339-L341)
Please let me know whether this is correct or not. It works.

bkueng, 2020-03-30:
@SalimTerryLi indeed, you're correct. Did you find more of these?
https://github.com/PX4/Firmware/pull/14528

SalimTerryLi, 2020-03-30:
> 
> 
> @SalimTerryLi indeed, you're correct. Did you find more of these?
> #14528

Sorry but I haven't done such work. We should do a complete inspection.

closed 2020-03-30 by dagar